ftctechnh/ftc_app #374: wifi direct does not

hadronknights, 2017-09-09:
With the latest ftc-app code 3.4 and with 3.5 rc app and 3.5 ds app, configuring wifi setting does not work. DS tries few times and gives up.

devsnek, 2017-09-09:
i had the same issue today and fixed it by disabling all wifi direct settings, clearing all saved connections for wifi direct, then doing the config only through the app, not android settings

hadronknights, 2017-09-10:
How do you setup using the app? Thank you

hadronknights, 2017-09-10:
We can connect now. Thank you @devsnek. But now we can't configuration. Edit option only allows us to scan but scanning is not working  even if   robot/rev expansion hub is connect and RC app is running. Any ideas?

WardBenjamin, 2017-09-11:
@hadronknights if the specific WiFi Direct problem is solved, please close this issue.

20avva, 2018-01-05:
@hadronknights check out #484. Sorry if I’m late

opdollar, 2019-11-21:
hello , found a solution to this, check that do not disturb.

closed 2019-11-21 by cmacfarl

opdollar, 2019-11-21:
never mind my team miss informed me.