ftctechnh/ftc_app #376: configure robot scan does not work

hadronknights, 2017-09-10:
In 3.4 scan under Configure Robot doesn't work. Moreover, it doesn't provide any way to edit the configuration manually in order to add the motors and sensors.

hadronknights, 2017-09-10:
btw, ftc RC app launches automatically when connected if the  REV expansion hub is already switched on but still scan does not work

gearsincorg, 2017-09-10:
Questions and suggestions:

1) How are you installing RC App?  (Android Studios or Play Store)
   If you are building with AS, make sure Instant Run is turned off (See
Tech forum for instructions)

2) What hardware DO you have connected (to help us diagnose)
  note, if you download the RC Log file, it will show what hardware it sees
during the scan.

3) If you just want to dummy something up to test your program, use the
"Configure from template" button at the bottom of the configure screen (do
not use the NEW button).
Pick a configuration close to what you have, or use the "everything" one (I
forget the exact name).
This way you can a least edit up something for your program.

Phil Malone
GEARS Inc.

phil.malone@mr-phil.com
Office: 301.387.2331
Mobile: 301.501.7424

hadronknights, 2017-09-11:
@gearsincorg please see below, any help appreciated it.
1) using app store and then deploy TeamCode from Android SDK
2) 4 motors, 1 encoder, of course and the Imu
3) Configure from template option does not give a template .. it states no template found.

BTW checked the version for RC and DS app and they indicate 3.4

gearsincorg, 2017-09-11:
OK, so you are seeing some big holes..

So, the big question you didn't answer... Have you turned off Instant Run
(in Android Studio)?

This "feature" seems to upset many of the FTC APP capabilities that depend
on file system access.
One of these capabilities is Templates,  the other is finding Opmodes to
display on the driver station..

To turn this off...

   1. Open the *Settings* or *Preferences* dialog.
   2. Navigate to *Build, Execution, Deployment* > *Instant Run*.
   3. Uncheck the box next to *Enable Instant Run*.

​Then I suggest you uninstall the current RC app, and then rebuild/deploy
it.

hadronknights, 2017-09-11:
We did disable the instant run but we will double check again. BTW this is log file we had. Initially, we had pairing issue, which was resolved.
[robotControllerLog-2.txt](https://github.com/ftctechnh/ftc_app/files/1292719/robotControllerLog-2.txt)

gearsincorg, 2017-09-11:
Well, it's definitely not seeing any of your deices...

09-09 18:36:07.502  9410 10675 I RobotCore: ========= Device Information
===================================================
09-09 18:36:07.502  9410 10675 I RobotCore: Type
                    Name                           Connection
09-09 18:36:07.502  9410 10675 I FtcEventLoop: ======= INIT FINISH =======

Normally this would be a list of all the USB devices.

Just to eliminate any Software Build issues, can you download the Play
Store RC app and run that to see if it has the same result.

Phil.

Phil Malone
GEARS Inc.

phil.malone@mr-phil.com
Office: 301.387.2331
Mobile: 301.501.7424

bflackgo, 2017-09-30:
We had the same issue.  We were able to fix it by deleting the Robot Controller app on the phone and then downloading it again from the Play Store.  Once reloaded, the REV hub showed up with a scan.  The software version was the same before & after - it didn't make sense but solved the problem for us.  Good luck.

NoahAndrews, 2017-09-30:
@bflackgo You probably had instant run turned on when you built the original RC app

closed 2017-12-03 by cmacfarl