ftctechnh/ftc_app #377: Change default text for opening PRs

qwertychouskie, 2017-09-10:
Right now it says:

`Before issuing a pull request, please see the contributing page.`

It seems this doesn't really work, see #375, #365, #362, etc.  It needs to get attention, something like:

`__STOP!__  If you are new to git and do not understand the mechanics of forks, branches, and pulls, then __do not__ push this button.  Most likely it won't do what you think it will.`

qwertychouskie, 2017-09-10:
ping @ftctechnh

brandon0x52, 2017-09-10:
This is a great idea IMO, I'd like to see less clutter on the Pull Requests page.

qwertychouskie, 2017-09-10:
And #378.  I almost feel like unsubscribing from this repo as every other email makes me want to facepalm.

Please @ftctechnh, save us!  Hear our cries of burden!  Be merciful to us!  Stop this onslaught of meaningless emails!

P.S. To anyone who has done this, it's really OK, it isn't clear and is quite understandable that this happens.  The issue here is the lack of clarity in the repo that causes the confusion.