ftctechnh/ftc_app #380: Webpage does not show embedded images

ibravo, 2017-09-11:
The Vuforia VuMarks wiki page does not show the embedded pictures when reaching it out via 
https://github.com/ftctechnh/ftc_app/wiki/Identifying-Vuforia-VuMarks

gearsincorg, 2017-09-11:
Thanks....  It looks like it's referencing images from a temporary test site.  Tom's been notified.

In the mean time you can get them here:

https://firstinspiresst01.blob.core.windows.net/ftc/vumark-us.pdf

ftctechnh, 2017-09-11:
Thanks ibravo.  I had placed them in a private repo prior to kickoff so the images wouldn't be released prior to kickoff.  In the excitement of kickoff on Saturday, I neglected to transfer the images to a public repo.

issued should be resolved now.

closed 2017-09-11 by ftctechnh