ftctechnh/ftc_app #381: RelicRecoveryVuMark and VuMarkInstanceId missing from FTC SDK

dominicap, 2017-09-11:
Both RelicRecoveryVuMark and VuMarkInstanceId seem to be missing from the latest FTC SDK (v3.4). Thus 'ConceptVuMarkIdentification' does not compile.
[image]

Any assistance would be appreciated.

gearsincorg, 2017-09-11:
> I do not see this...
>
> [image: image]
> <https://user-images.githubusercontent.com/1435639/30287129-c78658c4-96d8-11e7-83f7-9c8e74f0527d.png>
>
> I suspect that you are not linking against the 3.4 aar's. Can you describe
> exactly what process you followed to acquire the 3.4 SDK?

I concur.  I have built these samples from an unmodified (except for the
//@Disable line) FTC_APP zip download.

Note: If you have just downloaded, remember that you should "re-import" the
project.

Phil.

Phil Malone
GEARS Inc.

phil.malone@mr-phil.com
Office: 301.387.2331
Mobile: 301.501.7424

cmacfarl, 2017-09-11:
> Note: If you have just downloaded, remember that you should "re-import" the project.

Alternatively "Clean Project" followed by "Rebuild Project"

dominicap, 2017-09-11:
My bad, after re-cloning, the issue no longer persists. Thanks!

closed 2017-09-11 by dominicap