ftctechnh/ftc_app #382: Don't make Chrome the "recommended" browser

qwertychouskie, 2017-09-11:
1. Chrome requires Win 7+, meaning that teams that can't afford new computers can't use it
2. Chrome eats ram faster than you can say "Relic Recovery", again bad for people who can't afford newer systems
[image]
3. Why is there even a need for an "officially recommended browser"? Just say that it "works on Chrome, Firefox, Edge, Safari, and most other modern browsers.", and link to the installation instructions for each.

Making only one browser officially recommended, one that won't work for people that can't afford newer systems, seems... against the spirit of the project, making programming more accessible to everyone.  People should select a browser based on what fits their needs, on a comparison of things like security, reliability, privacy, speed, whether it works on their system, etc., rather than a web app saying that a certain single browser is "recommended" for something when many other browsers work just as well, and often even better for some.

Ping @ftctech

WardBenjamin, 2017-09-11:
An "officially recommended browser" ensures that the FIRST team can focus their efforts on one browser instead of needing to test and support a bunch. If it works on other platforms, great, but ensuring that every detail works is a waste of their time. (Modern) Chrome really isn't that bad about RAM usage if you're not going to have a ton of tabs open, and it's probably the most common browser on Windows. If   a team is using a pre-Windows 7 computer, which is not true in the _VAST_ majority of cases because XP and Vista are both very old and unsupported operating systems anyway. If the OS isn't supported by _Microsoft_, then there's no real reason to cover that very fringe edge case.

cmcknight, 2017-09-11:
What about "web standards?" Unless the devs are using Chrome-specific extensions, the app should work on any modern browser. It is reasonable to specify a particular minimum version to minimize the number of browsers that need to be tested. However, I'd be more hesitant to specify a single browser that may or may not work equally well across all platforms. My team uses a mix of Windows (7, 8 & 10), MacOS, and Linux on their laptops and while Chrome is available on all of those it does behave slightly differently on each OS and each OS version.

Is the dev team using tools like Selenium and Jenkins for test automation and regression testing? Test automation would relieve some of the burden of testing and likely produce faster turnaround times as well as making it immediately obvious when something breaks.

Just my tuppence.

qwertychouskie, 2017-09-12:
@cmcknight Totally agree.  I think that was what I was trying to say without knowing it was what I was trying to say ;)

gearsincorg, 2017-09-12:
Playing devils advocate here.....

Since all the current development and pre-testing is done by the FTC Tech
Team, who are all volunteers, baring Tom, perhaps someone in the community
would like to volunteer to do the browser compatibility testing discussed
here.

Those testing tools sound cool, but they carry their own learning curve.
Perhaps someone with experience would like to configure a test suite.

Since it would be testing the exposed browser interfaces, could the testing
could be done independently of the underlying development environment.
 Thus, no special "permissions" would be required.

In a round-about way, I'm trying to make the point that everyone's
resources are limited, and if you think something is super important, then
perhaps you could find a way to help make it happen.  Asking for it, is one
way....

Just my 10c worth.

Phil.

cmcknight, 2017-09-12:
Hi Phil,

You’re spot on about everyone’s resources being limited. However, I’ve got a few bits of spare time to take a whack at it. I’ve set up a Jenkins/Selenium test environment in a VMWare virtual machine running Ubuntu Server 17.04 and gradle is in the process of downloading half of the Internet (grin) so that I can compile things up. My question is whether or not there are any specs/requirements/etc. that a test suite can be written against. Could you point me to where that information might be?

Thanks!

Chuck

gearsincorg, 2017-09-12:
Hi

Since I don't know how these test suites work, I'm not sure what sort of
specs/requirements are required.

I would assume that for us, the key "compatibility" issues would relate to
the correct formation of the screens (frames, menus, popups)and the support
for any and all of the Java script commands used by the pages.

Is there a way to run the raw pages through a "universal checker" ?

Seems like if I was building a validation suite, that's the first tool I'd
create.

Phil.

Phil Malone
GEARS Inc.

phil.malone@mr-phil.com
Office: 301.387.2331
Mobile: 301.501.7424

cmcknight, 2017-09-12:
Hi Phil,

Selenium automates the keypresses/button clicks for browsers and is typically scrript-driven. There are also ways to capture screen shots and validate that the screens are displaying and working as designed. Jenkins just ensures that the software is automatically rebuilt when a commit is made to the repository and rats out whoever broke the build with a polite “please fix it” message. OK, the message is customizable but we are gracious professionals so we’ll use a polite message. :-)

The specs/requirements that I’d need would involve both the functional (what happens when some event occurs) and the non-functional (what should the screen display, performance, etc.). The rest is a matter of writing the scripts to run the test scenarios and report on the outcomes. The challenge will be in having to reverse-engineer the specs/requirements/etc. if they don’t exist now. The major challenge with anything that is browser-based is the screen size and how the UI handles that. There has been a lot of work in that area in the web end of things to create UIs that adapt to different screen sizes that is worth looking at, but it’s not a panacea for all cases.

Regards,

Chuck

ftctechnh, 2017-09-12:
Hi Folks - Chrome is recommended but not required.  We have done tests (not super extensive) with other browsers (IE, Edge, FireFox, Safari, etc.), but have done the most testing using Chrome (on multiple platforms, including different versions of Windows, MacOS and Android).

closed 2017-09-12 by ftctechnh

LBYPatrick, 2017-10-17:
They recommended chrome because it uses **Chromium**, a modern browser core. And I don't think there are teams that are too poor to buy computers with Win 7+ since Android Studio actually has high requirements on computer hardware specs than chrome. If you want to use Firefox, go ahead (Oh Microsoft Edge requires Windows 10, that would be more ridiculous)

qwertychouskie, 2017-10-17:
@LBYPatrick If they were recommending based on the most modern browser core, they would have recommended Servo ;)

P.S. Since Servo is being integrated into Gecko over the next few versions (Stylo in 57, maybe Webrender in 58), soon Firefox will have the most modern engine of the mainstream browsers :D
:fox_face: