ftctechnh/ftc_app #383: OnBot Java Instructions only provided for Windows 10

qwertychouskie, 2017-09-11:
According to https://www.netmarketshare.com/operating-system-market-share.aspx?qprid=10&qpcustomd=0, Windows 7 is WAY more common than 10.  Despite this, https://github.com/ftctechnh/ftc_app/wiki/Connecting-a-Laptop-to-the-Program-&-Manage-Network only provides instructions for Windows 10.  Also, the page makes it seem like Windows 10 is the only option, as no other systems are even mentioned.

I am willing to help document other systems if wanted, just let me know.

Ping @ftctechnh

WardBenjamin, 2017-09-11:
If you would like to add documentation for other systems, please do. However, the FIRST team doesn't have unlimited time, and Windows 10 is what the majority of new teams use anyway (since new laptops ship with Windows 10). If there's a veteran team that hasn't figured it out by now, there's a bigger issue.

qwertychouskie, 2017-09-11:
@WardBenjamin The wiki is non-editable, so @ftctechnh would have to give access.

> Windows 10 is what the majority of new teams use anyway (since new laptops ship with Windows 10).

That assumes that the majority of new teams can afford to buy new laptops (which I doubt based on the stats linked to in the first post).

WardBenjamin, 2017-09-11:
@qwertychouskie you can write documentation in the same form in another repository, a fork, or even in this issue. Same syntax.

The majority of teams, at least in my region, already have access to a Windows 10 device through either their members, their member's parents or mentors, their school, etc etc etc.

magneticflux-, 2017-09-11:
Honestly, if teams can't figure out how to translate Windows 10 instructions to Windows 7 menus, they're going to have trouble getting anything working. I don't really see an issue with providing Windows 10-only tutorials since the menus have changed very little (besides the new theme) since Windows 7. Additionally, the teams must know how to set up wifi to even use their laptops unless they only have Ethernet access for some reason.

ftctechnh, 2017-09-12:
Hi qwertychouskie,

Thanks for the feedback... Yes, the documentation only provides steps on how to connect using a Windows 10 machine.  however, the steps required for a Windows 7 machine, or even for a Chromebook, MacBook or Android device, are analogous and the Windows 10 instructions serve as a good model for how to connect.  A user can consult their device's documentation for details on how to connect to a Wi-Fi network (which is what is required to access the OnBot Java or Blocks Programming tools.

closed 2017-09-12 by ftctechnh