ftctechnh/ftc_app #389: Can't build unmodified app

JBYoshi, 2017-09-14:
When I run `./gradlew build` on the unmodified project (commit 9f1e829930e7e5e9c74d26aada5e19742dcd1e82, the latest at the moment), I get this error:

```
(...)
:FtcRobotController:bundleRelease
:FtcRobotController:compileReleaseSources
:FtcRobotController:assembleRelease
:FtcRobotController:assemble
:FtcRobotController:lint
Ran lint on variant debug: 16 issues found
Ran lint on variant release: 16 issues found
Wrote HTML report to file:///home/jbyoshi/git/CDARobotics/FtcRobotController/build/reports/lint-results.html
Wrote XML report to file:///home/jbyoshi/git/CDARobotics/FtcRobotController/build/reports/lint-results.xml
:FtcRobotController:lint FAILED

FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':FtcRobotController:lint'.
> Lint found errors in the project; aborting build.
  
  Fix the issues identified by lint, or add the following to your build script to proceed with errors:
  ...
  android {
      lintOptions {
          abortOnError false
      }
  }
  ...

* Try:
Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output.

BUILD FAILED

Total time: 7.939 secs
```

The specified lint files are:

[lint-results.html.txt](https://github.com/ftctechnh/ftc_app/files/1304718/lint-results.html)
[lint-results.xml.txt](https://github.com/ftctechnh/ftc_app/files/1304719/lint-results.xml)

Why is this happening?

magneticflux-, 2017-09-14:
Can you run `clean` and then `build`? After that, can you also reimport the project in Android Studio? It seems like your JDK might not be set up properly as well.

JBYoshi, 2017-09-14:
After looking into it, the `assemble` task should work fine for my usage.

The root cause seems to be that RobotCore.aar has a copy of `javac` in it, and never got the NIO files stripped (NIO isn't available on Android).