ftctechnh/ftc_app #392: Encountering a ConcurrentModificationException when @OpModeRegistrar is used on more than one method

stormymcstorm, 2017-09-16:
I'm having trouble using the OpModeRegistrar annotation. While trying to use it I get a ConcurrentModificationException. While trying to find the error I tried removing the OpModeRegistrar annotation from [ConceptRegisterOpModes](https://github.com/ftctechnh/ftc_app/blob/master/FtcRobotController/src/main/java/org/firstinspires/ftc/robotcontroller/external/samples/ConceptRegisterOpModes.java) and everything worked as expected. This leads me to believe that this error occurs because there is more than one method using the OpModeRegistrar annotation. [This is the stack trace for the ConcurrentModificationException](https://gist.github.com/stormymcstorm/fc3773ef146250c49835e9f988f839ab).

Steps for reproducing
1. Make a copy of [ConceptRegisterOpModes](https://github.com/ftctechnh/ftc_app/blob/master/FtcRobotController/src/main/java/org/firstinspires/ftc/robotcontroller/external/samples/ConceptRegisterOpModes.java) in your TeamCode directory (rename if you see fit)
2. Install and run the app. Notice how the robot fails to start
3. View the app logs. This is easiest done using Android Studio's Android Monitor 
4. Comment out the line in [ConceptRegisterOpModes](https://github.com/ftctechnh/ftc_app/blob/master/FtcRobotController/src/main/java/org/firstinspires/ftc/robotcontroller/external/samples/ConceptRegisterOpModes.java) containing `@OpModeRegistrar`
5. Install and run the app. Notice how the app works just fine now 
6. Make another copy of  [ConceptRegisterOpModes](https://github.com/ftctechnh/ftc_app/blob/master/FtcRobotController/src/main/java/org/firstinspires/ftc/robotcontroller/external/samples/ConceptRegisterOpModes.java) in your TeamCode directory and name it something different
7. Install and run the app. Notice how the robot fails to start. This shows that the problem isn't that the OpModeRegistrar is in the FtcRobotController and that is occurs when any two methods are annotated with OpModeRegistrar

gearsincorg, 2017-09-17:
Thanks for the detailed analysis.

This used to work, so we'll need to see what's changed.

Phil.

Phil Malone
GEARS Inc.

phil.malone@mr-phil.com
Office: 301.387.2331
Mobile: 301.501.7424

cmacfarl, 2017-09-18:
Following the steps outlined by @stormymcstorm I can not reproduce this.  @gearsincorg you have any luck?

gearsincorg, 2017-09-18:
Hi

In my last correspondence I described the varius items needed for SuGO.

Not all of these items are available from the SuGObot.com store.  Some are
from LEGO (as indicated in the original email).

If you need to get actual SuGObot.com prices for a purchase, please just
visit the online store (as indicated in the email)

http://sugobot.com/store.htm

If you can't purchase online, then just build an order online (which will
include shipping) and then simply print out the Shopping cart and use this
as a quote.

Thanks for your interest in SuGO.

Phil..

Phil Malone
GEARS Inc.

phil.malone@mr-phil.com
Office: 301.387.2331
Mobile: 301.501.7424

magneticflux-, 2017-09-18:
@cmacfarl Are you sure it wasn't just a fluke? Concurrency issues are notoriously hard to diagnose and reproduce since they could depend on the speed of the device or other complex factors. It's technically possible you just got really unlucky.

cmacfarl, 2017-09-18:
Well I'm humble enough to never say never, but I don't believe that this is actually a multiple thread concurrency issue, but rather a modification of a collection while iterating over same.

Note this text here: https://docs.oracle.com/javase/7/docs/api/java/util/ConcurrentModificationException.html

>  If a single thread issues a sequence of method invocations that violates the contract of an object, the object may throw this exception. For example, if a thread modifies a collection directly while it is iterating over the collection with a fail-fast iterator, the iterator will throw this exception. 

The fact that @stormymcstorm was able to reproduce the problem at will and describe it so succinctly indicates that there's a problem _somewhere_, we just don't yet know what the difference between @stormymcstorm's setup is and mine.

@stormymcstorm are you using 3.4?

stormymcstorm, 2017-09-18:
@cmacfarl Yes I am using 3.4. I just tried to reproduce it again myself, and it's no longer throwing. I'm try to figure out what I changed to cause this.

cmacfarl, 2017-09-19:
If it happens again can you please post the full log someplace accessible?