ftctechnh/ftc_app #395: Driver Station cannot find any of our OpModes

uridvir, 2017-09-17:
We have two Op Mode classes in our TeamCode folder: TeleOpDrive and VuforiaTest. TeleOpDrive was written by us at @8148alephbots, and VuforiaTest is a copy of the file ConceptVuMarkIdentification.java from external.samples. All we did was rename it, put in our Vuforia product key, and comment out the line with "@Disabled".

When we uploaded the code, trying to select the Op Mode on the Driver Station yields this:
[image]

Here is our GitHub repository, please use the 'debug' branch: https://github.com/8148alephbots/relic-recovery.git

When starting the app after building, it throws an IO exception related to OnBotJava. Here is the log after building and running the Robot Controller app:
[relic-recovery-log.txt](https://github.com/ftctechnh/ftc_app/files/1309291/relic-recovery-log.txt)

dchotzen-hartzell, 2017-09-17:
My team was having this problem yesterday. The solution is to disable Instant Run on Android Studio, as it doesn't properly upload the code to the phone. 

Go to settings/preferences in Android Studio > Build, Execution, Deployment > Instant Run, and uncheck "Enable Instant Run to hot swap code/resource changes on deploy," and the issue should be fixed.

magneticflux-, 2017-09-17:
@uridvir Like @dchotzen-hartzell said, Instant Run is likely the culprit.

FYI, sharing your GitHub URL doesn't allow anyone to add to your debug branch. It looks like you think that is so based on the latest commit message.

uridvir, 2017-09-18:
Sorry, by "use" I meant that if needed, you should pull from the "debug" branch to test the issue. I know you can't push to the repo without write access.

cuiwanj, 2017-11-11:
if I don't choose the Instant Run,I can't download the APP.

magneticflux-, 2017-11-11:
@cuiwanj You'll have to be more specific than that. How are you running the app?

cuiwanj, 2017-11-11:
@magneticflux-  I'm running the app on Android studio.

magneticflux-, 2017-11-11:
@cuiwanj Could you provide a screenshot of what you see? I'm not aware of any situations that would only allow Instant Run to be used. You can take an image using the Snipping tool in Windows or shift-command-4 on Mac and upload it by dragging it to the comment box.

cuiwanj, 2017-11-11:
@magneticflux- [image]
Thank you very much!But I assure that no APP on my phone.

cuiwanj, 2017-11-11:
@magneticflux- If I choose the INSTANT RUN,I can run the APP on Android studio.So there's on program as the same as uridvir !

magneticflux-, 2017-11-11:
@cuiwanj To install apps directly from Android Studio, you must enable the "Install via USB" option in the Developer Settings menu of the phone. Once you have done that, the phone should *not* cancel the installation.

cuiwanj, 2017-11-11:
@magneticflux- thank you!I think I know the reason,but my phone is xiaomi ,I can't do that like you! Thank you very much!

magneticflux-, 2017-11-11:
@cuiwanj It looks like you have to use these special instructions [here](https://stackoverflow.com/questions/37641670/adb-install-failure-install-canceled-by-user) because it's a Xiaomi phone.

Glad I could help!

closed 2017-12-03 by cmacfarl