ftctechnh/ftc_app #397: DbgLog.class missing from latest release

Timberline10435, 2017-09-19:
With newest code, our programs no longer compile because DbgLog isn't resolved.  Looking at classes.jar from the zip file, I see it is no longer there.

cmacfarl, 2017-09-19:
DbgLog was removed as it duplicated RobotLog functionality and it was deemed that we didn't need both.  RobotLog.d() is likely what you want.

Timberline10435, 2017-09-19:
Thanks for the info.  Can you add that to the Readme for 3.4?  Where might we find documentation of RobotLog?  We see it's in com.qualcomm.robotcore.util.RobotLog.

ftctechnh, 2017-09-22:
@Timberline10435 
Great suggestion.  Will do (update readme to reflect DbgLog removal).

markshancock, 2017-09-22:
This change caused our team some initial startup challenges.
It would have helped us if you had deprecated the class rather than deleting it.
http://docs.oracle.com/javase/7/docs/technotes/guides/javadoc/deprecation/deprecation.html

closed 2017-12-03 by cmacfarl