ftctechnh/ftc_app #399: Cannot override LinearOpMode.init();

20esaua, 2017-09-22:
I know that `init()` in `LinearOpMode` has been declared as `final`, but I need to be able to execute a few functions on initialization.

Would it be okay to put the code in the constructor? The code will do a few things, including:
* Initialize `DcMotor`, `ColorSensor`, `GyroSensor`, and a few other objects
* Calibrate gyro sensor
* Print debug info
* Display a message letting the users know that the opmode has been initialized
* And perhaps a few other things in the future...

If the LinearOpMode's constructor is called when the app starts, the code wouldn't work in the constructor. But if the constructor is called when the user selects the opmode and taps the init button, it should work.

Any ideas?

20esaua, 2017-09-22:
Ah, and I also need to override `LinearOpMode.loop()` too.

qwertychouskie, 2017-09-22:
It seems the proper way to do this is
```
public void runOpMode() {
...init stuff...
waitForStart();
...do stuff...
}
```
For more detail see https://github.com/FTCTeam10298/2016-17-code/blob/master/TeamCode/src/main/java/org/firstinspires/ftc/teamcode/Auto.java

20esaua, 2017-09-22:
Thanks a lot @qwertychouskie. That should do it. Also, for future reference: https://ftc-tricks.com/overview-linear-opmode/ :+1:

closed 2017-09-22 by 20esaua

gearsincorg, 2017-09-22:
> overview-linear-opmode/ 👍

Just in case it's not obvious.....

If you want to use the init() and loop() methods, just base your op-mode on
the vanilla OpMode class, rather than the LinearOpMode class.

Phil.

Phil Malone
GEARS Inc.

phil.malone@mr-phil.com
Office: 301.387.2331
Mobile: 301.501.7424