ftctechnh/ftc_app #400: Modern Robotics Range sensor with Rev hub causes the program to hang

FestiveInvader, 2017-09-23:
I was trying to use the Modern Robotics Range sensor with the Rev module, and the program stops whenever I try to use the range sensor.  Replacing the MR range with a Rev Color/Distance fixed the problem, but I'd prefer to use the longer range Range.

qwertychouskie, 2017-09-23:
For the sensors, you have to use an additional adapter wire along with the level converter.  Do you have this plugged in?

FestiveInvader, 2017-09-23:
Yes, I have the crossover cable along with the logic level converter plugged in, and have tried flipping different connections around in case one was wrong. 

Note that not only was my code not working, but also the sample opmode. 

Thanks,
-Eric

magneticflux-, 2017-09-23:
@FestiveInvader I'm not certain, but flipping the connectors around may have damaged the equipment. It probably *shouldn't*, but I don't know the exact specs of the sensor and I've heard of it happening to other things.

FestiveInvader, 2017-09-23:
We have multiple sensors, as well as multiple Rev modules. I did refer to the wiring diagram to get the proper wiring the first time, but after that didn't work I flipped it. I have tried all three range sensors we have and none of them work. I haven't tried to use them with the Modern Robotics CDIM again, but as of last testing (mid July) the Range still worked. That was on v3.2 if I recall correctly. 

Thanks,
-Eric

ftctechnh, 2017-09-28:
Eric,

Could you post a copy of your op mode and also maybe a copy of the log file and reference the date and time when the hang/problem occurred?

Tom

FestiveInvader, 2017-10-27:
Sorry for long hiatus, the season started and all that.  Turns out that I was referencing a different sensor in my program, and thought it was a hardware problem.  I'm sorry to take up your time, and wish you a good season!

closed 2017-10-28 by cmacfarl