ftctechnh/ftc_app #402: VuMark asset size larger than printed sheet

haithcoc, 2017-09-25:
The VuMark sizes specified in the RelicVuMark.xml asset (304.8mm x 223.63mm = 12" x 8.8") are greater than the sheet size (8.5" x 11") given in the instructions.
It would seem that having sizes different from the printed images would cause issues in the transformation matrices that feed the VuforiaLocalizer results.
What should the config (xml) values be?
What should the margin be on an 8.5"x11" sheet when the images are correctly printed?
Are these margins to the boundary created by the green leaves, or do they include the portions of the FTC logo that extend beyond the rest of the image?
Finally, when the images are placed on the field, Figure F-6 in part 2 of the manual shows 3 inches from the line the tile seam/balancing stone center/jewel holder.  Is the 3 inches to the edge of the paper, or to the margin on the image?

Thanks

ftctechnh, 2017-09-28:
Hi haithcoc,

I believe that you want to print the VuMarks using the appropriate PDF files and make sure scaling is disabled when you print the marks.

If you are using an 8.5" x 11" piece of paper, use the "Pictograph US" version and print it as is (make sure scaling is disabled).

https://www.firstinspires.org/sites/default/files/uploads/resource_library/ftc/pictograph-us.pdf

As for the size of the asset, I believe that the most important parts of the VuMark image are the area bounded by the orange and black borders, plus the orange and black hexagons that encode the data in the VuMark.  Even though the actual asset size might be larger, the relevant images fit within the 8.5 x 11 sheet of paper.  

If you print the image without scaling, you should be good to go.

haithcoc, 2017-09-28:
I have printed the VuMarks from the PDF files with scaling disabled (actual size/scale 100%) in landscape.
And I realize that "the most important parts of the VuMark image are the area bounded by the orange and black borders, plus the orange and black hexagons that encode the data".  However, a VuMark is a Trackable/ImageTarget.  Therefore, they still support the interface to get trackable pose, and therefore determine camera/bot position/orientation relative to the image.
If I understand things correctly, the asset size will effect the trackable to camera transformation matrix.
It would seem that this matrix will not be accurate if the input sizing information is not accurate.
Is there a reason that the asset xml has the width at 12"?  Does the image data in the asset .dat file match the images in the pdf (without the hexagon encoders)?  Or are does the image data contain some kind of border that will make the 12"x8.8" specified size result in correct transformations?

ftctechnh, 2017-09-28:
Hi haithcoc,

I think you should be OK...  I believe the vuforia software calculates the pose information based on the visible portions of the VuMark.  

You can test this by running the demo op mode and looking at the pose data and comparing it to what you actually measure in your test environment.  

Tom

haithcoc, 2017-10-09:
We still need to compare actual vs. "Vuforia computed from trackable pose" camera location.
We will post results when we do.
However, it would still help to know what the printed image size should be.
Different printer defaults and different pdf viewing software (Acrobat, browser plugin, google), may behave differently with respect to print scaling.
Further, the borders on screen to not look like those in the printed image.
Can someone please provide the proper dimensions of the printed image - width/height and distance from page edge of the green leaf portion would probably be easiest to measure?
We are hosting a qualifier, and want to make sure we are setting the field up properly.

ftctechnh, 2017-10-10:
@haithcoc 
Hi haithcoc,

We printed out the official US Letter version of this season's VuMark without any scaling applied using Adobe Acrobat ("actual size" option).  The base of the bottom black border is 6.625" when we do this (see image below for dimension):

[image]

haithcoc, 2017-10-10:
Awesome.  Thanks.  Still not sure why the xml dimensions are 12"x8.8", and I still need to confirm trackable-based camera position works, but everything else is good.  Closing.

closed 2017-10-10 by haithcoc