ftctechnh/ftc_app #407: ColorNumber

archishou, 2017-10-07:
WHY HAVE YOU STILL NOT NATIVELY ADDED COLOR NUMBER. Color is far more accurate than the method you have included for this year.

gearsincorg, 2017-10-07:
Who are you directing this at?   
You don't mention the specific sensor you are referring to, or who you expect to make the change.  (ie: the sensor vendor, the FTC Development team or some third party. )

More importantly, why don't you provide an example of what you are proposing.  
I'm unsure exactly what you mean by "Color Number" and "Native" in this instance.

Phil.

archishou, 2017-10-07:
Im directing this to whoever made the SDK, i'm not sure who. The sensor i'm talking about is the Modern Robotics Color Sensor color number is a feature that is way better than the raw red, blue, white, green i2c register readings. When i say native i am referring to the sdk, it is not a native feature where i can pull the color number like colorsensor.red(); and so forth.

gearsincorg, 2017-10-07:
OK, so in looking at the Modern Robotics sensor I see that it has a
processed "Color Number" to indicate the likely color.

I'm not sure why this wasn't included in the API, although it may be
because most other sensors do not have this color estimate.
It's probably there to emulate the LEGO color sensor that it replaced.

It would certainly be easier to use in your program, but I don't think
calling it "far more accurate" is reasonable.
It's an attempt to characterize the basic color, but if you are looking to
distinguish between two colors (like in this year's game) it's much more
accurate to use the pure red and blue sensor data.

I'll pass on your "request", on the basis that it's simpler to use.

Phil.

archishou, 2017-10-07:
No, I have written an i2c driver that interacts with the color sensors so i can get color number and yes, it is far more accurate

archishou, 2017-10-07:
Last year, I would like to say I say an around 33-45% increase in accuracy, if that means anything