ftctechnh/ftc_app #411: Error starting robot: java.lang.NullPointerException: Attempt to invoke virtual method 'byte[] com.qualcomm.hardware.lynx.commands.LynxDatagram.toByteArray()' on a null object reference

JBYoshi, 2017-10-11:
```
java.lang.NullPointerException: Attempt to invoke virtual method 'byte[] com.qualcomm.hardware.lynx.commands.LynxDatagram.toByteArray()' on a null object reference
     at com.qualcomm.hardware.lynx.LynxUsbDeviceImpl.transmit(LynxUsbDeviceImpl.java:745)
     at com.qualcomm.hardware.lynx.LynxUsbDeviceDelegate.transmit(LynxUsbDeviceDelegate.java:186)
     at com.qualcomm.hardware.lynx.LynxModule.retransmit(LynxModule.java:1161)
     at com.qualcomm.hardware.lynx.commands.LynxRespondable.awaitAndRetransmit(LynxRespondable.java:364)
     at com.qualcomm.hardware.lynx.commands.LynxRespondable.awaitAckResponseOrNack(LynxRespondable.java:372)
     at com.qualcomm.hardware.lynx.commands.LynxRespondable.sendReceive(LynxRespondable.java:253)
     at com.qualcomm.hardware.lynx.LynxDcMotorController.internalGetMotorChannelMode(LynxDcMotorController.java:391)
     at com.qualcomm.hardware.lynx.LynxDcMotorController.internalSetZeroPowerBehavior(LynxDcMotorController.java:442)
     at com.qualcomm.hardware.lynx.LynxDcMotorController.setMotorPowerFloat(LynxDcMotorController.java:591)
     at com.qualcomm.hardware.lynx.LynxDcMotorController.floatHardware(LynxDcMotorController.java:758)
     at com.qualcomm.hardware.lynx.LynxDcMotorController.initializeHardware(LynxDcMotorController.java:141)
     at com.qualcomm.hardware.lynx.LynxController.initializeHardwareIfNecessary(LynxController.java:221)
     at com.qualcomm.hardware.lynx.LynxController.hook(LynxController.java:332)
     at com.qualcomm.hardware.lynx.LynxController.adjustHookingToMatchEngagement(LynxController.java:319)
     at com.qualcomm.hardware.lynx.LynxController.moduleNowArmedOrPretending(LynxController.java:120)
     at com.qualcomm.hardware.lynx.LynxController.finishConstruction(LynxController.java:92)
     at com.qualcomm.hardware.lynx.LynxDcMotorController.<init>(LynxDcMotorController.java:136)
     at com.qualcomm.hardware.HardwareFactory.mapLynxModuleComponents(HardwareFactory.java:766)
     at com.qualcomm.hardware.HardwareFactory.mapLynxUsbDevice(HardwareFactory.java:714)
     at com.qualcomm.hardware.HardwareFactory.createHardwareMap(HardwareFactory.java:185)
     at com.qualcomm.ftccommon.FtcEventLoopHandler.getHardwareMap(FtcEventLoopHandler.java:127)
     at com.qualcomm.ftccommon.FtcEventLoop.init(FtcEventLoop.java:180)
     at com.qualcomm.robotcore.eventloop.EventLoopManager.startEventLoop(EventLoopManager.java:540)
     at com.qualcomm.robotcore.eventloop.EventLoopManager.setEventLoop(EventLoopManager.java:514)
     at com.qualcomm.robotcore.eventloop.EventLoopManager.start(EventLoopManager.java:456)
     at com.qualcomm.robotcore.robot.Robot.start(Robot.java:59)
     at com.qualcomm.ftccommon.FtcRobotControllerService$RobotSetupRunnable.startRobot(FtcRobotControllerService.java:259)
     at com.qualcomm.ftccommon.FtcRobotControllerService$RobotSetupRunnable$1.run(FtcRobotControllerService.java:276)
     at com.qualcomm.robotcore.util.ThreadPool.logThreadLifeCycle(ThreadPool.java:709)
     at com.qualcomm.ftccommon.FtcRobotControllerService$RobotSetupRunnable.run(FtcRobotControllerService.java:267)
     at java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:423)
     at java.util.concurrent.FutureTask.run(FutureTask.java:237)
     at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1113)
     at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:588)
     at com.qualcomm.robotcore.util.ThreadPool$ThreadFactoryImpl$1.run(ThreadPool.java:764)
     at java.lang.Thread.run(Thread.java:818)
```

Full logs: https://gist.github.com/JBYoshi/ea4fdff679ce0b059121abe2ade2a1cd

magneticflux-, 2017-10-11:
@JBYoshi Have you been able to reproduce the error with Instant Run disabled?

JBYoshi, 2017-10-12:
I did have Instant Run disabled. However, the issue seems to be fixed now. I'm guessing someone just forgot to recharge the battery.

closed 2017-10-12 by JBYoshi

magneticflux-, 2017-10-12:
@JBYoshi Have I seen you around the MCForge/Sponge area of GitHub before? Your username/icon looks familiar to me somehow.

JBYoshi, 2017-10-12:
Yep 😉

NoahAndrews, 2017-11-03:
I just got this error on the Google Play version of the RC. It's not an Instant Run issue.

NoahAndrews, 2017-11-03:
I'm just spitballing here, but we saw the hub that the RC was connected to get into a weird state. It had power from both the phone and the 12V battery, but the light was off. The secondary hub connected to that one had a solid yellow light, which may indicate it was being scanned via RS485. I think this, at its core, may actually be a firmware bug with the Expansion Hub.

NoahAndrews, 2017-11-03:
Another thing was that before showing the NPE, the robot would spend an unusually large amount of time in the "initializing, starting robot" state. What could cause that?

NoahAndrews, 2017-11-03:
@JBYoshi, I'd appreciate it if you'd reopen this issue. It went away for me too, but it's definitely a bug, and I think it's worth trying to reproduce.

gearsincorg, 2017-11-04:
Can whoever is seeing this error please describe the robot configuration so
we can establish the conditions under which it occurs?

the usual stuff....

Phone, OS,
Programming language
Hardware configuration
Program in question
Factors leading up to error
Visible indications (leds, screens etc.)

Thanks.

Phil Malone
GEARS Inc.

phil.malone@mr-phil.com
Office: 301.387.2331
Mobile: 301.501.7424

NoahAndrews, 2017-11-04:
Of course, sorry.

I experienced the issue on a Moto G4 Play running 6.0.1 (May security patch). The team is using Blocks programming. The app installed at the time I kept getting the error was version 3.5 from Google Play, but I had just uninstalled version 3.4 (official version, built from Android Studio).

[Link to configuration and robot log](https://gist.github.com/NoahAndrews/280d019d42a4da966245789d5764e30a). There's not really a "program in question". The error occurred while the app was setting things up, before you could select an OpMode. 

Looking at the log and the corresponding code, the exact points where the problem occurs are whenever the module attempts to tell the motors to stop as it populates the `hardwareMap`, and when a `LynxFailSafeCommand` is sent from `DefaultOpMode`. These commands appear to send without throwing an exception (I believe indicating that the `serialization` field is set at that point), but no response is received, and when a retransmission is attempted, the message's `serialization` field is not set for some reason, causing the exception. Further down, I found similar problems with other communications.

[I added some logging to OpenFTC to at least determine whether the serialization field is being unset somehow, or whether it was never initialized properly when the commands initially send.](https://github.com/OpenFTC/OpenFTC-app/commit/edee1e1f8c72a748df8730d1732a9125ed013dcd) It would be great if someone using our version encounters this issue and is able to provide a very slightly more helpful log. Unfortunately, the robot stopped showing the issue, even before I could install the app with enhanced logging onto the RC. Thus far, I have been unable to reproduce. :( If you can think of any additional logging that would be useful in diagnosing this, I'd love to have it added to OpenFTC. I think this is one area where OpenFTC's much quicker release process can allow us to benefit the FTC Tech Team, especially if we get a decently large userbase.

As stated, there was one point where the REV hub connected via USB didn't have its LED on at all, while the hub connected to that one via RS485 showed a yellow light. We got the error even when the hubs were not in that state, however.

[image]

gearsincorg, 2017-11-04:
Thanks, that's more helpful.  
Hopefully @cmacfarl or @rgatkinson can make something out of the failure sequence.

Once the fault occurred once, did it continue to occur?
if not, what actions did you take to make it go away?
What did you try that did not make it go away?

NoahAndrews, 2017-11-06:
It did continue to occur, until we left the robot off for a while and came back to it. Various combinations of power cycling the robot, choosing "restart robot" and restarting the apps (including swiping them out of the app switcher) failed to resolve the issue.

wxtrav, 2017-11-12:
One of our teams ran into this exact same thing today. Power cycles of robot, restarting app (via swiping them out), did not fix it.  Full power off of phones and robot both did, but that took a while (and was at the start of a match in one of the cases).

This was with G4 Play (verizon prepaid edition), Android Studio (Instant Run off), 3.5 SDK, 1.7 REV firmware. Exact error as on the screen above.

gearsincorg, 2017-11-12:
I've seen some "similar" issues since I've been working with two Control Hubs.

At some point, communications between the two hubs (and the phone) break down, and it seems to me to be related to disconnecting the phone when the hubs are powered, or possibly powering down the Hubs, while the phone is still connected.

At the point where the issue occurs, we go through a simple (but strict recovery process).
The key part is that ALL power needs to be removed from the control hubs.  This means Turning OFF the 12V AND powering down, or disconnecting, the phone.  Then the App is swiped off (to remove it from memory), the Hubs are powered back up, and then the phone is reconnected.

Since we often use Andriod Studio (and we need to unplug the phone to download code), we now make sure that whenever we unplug the phone, we ALSO power down the Hubs.  Ensuring that we then power them back up again before reconnecting the phone.

This procedure then ensures that all the devices start out clean.

Once in a rare while, this process does not work, so we then add the extra step of Restarting the Phone while the Hub power is off and USB is disconnected.

wxtrav, 2017-11-12:
Hmmm.... Thanks for the pointers. They were running at a competition yesterday, and were unplugging the phones to charge them and plugging them back in.  I am not sure what order this happened in, but that give us something to log.  It happened twice before a match (once they just sat there on the balance board, and the second time got it running after power cycling the phone and robot).

They usually don't unplug the phones when developing with Android Studio since they use the wireless adb (connected to the Wifi Direct).  Prior to updating the firmware to 1.7 a couple of days ago it was just hanging the app and not generating the error.

Two other possibly related things are:
(1) The NavX that is plugged into the 2nd hub (not the one connected via USB) lost connection as well (error is "problem with NavX").  We can test that it's not a communication issue between the hubs by connecting it to the first hub.
(2) one of our other teams has an issue where the two control hubs won't talk to each other at all. They are each recognized individually, but we swapped out multiple cables and it's like the port is just bad.

I'm going to have the teams start logging every power cycle or unplug / replug event, and will report what we find.

cmacfarl, 2017-11-12:
Folks, if you see this problem, please immediately gather the logs on both
the robot controller and driver station and attach them to this thread.

Thank you.

wxtrav, 2017-11-12:
The times stamps are way off, but these are from yesterday afternoon and the day before.

[robotControllerLog.txt](https://github.com/ftctechnh/ftc_app/files/1464950/robotControllerLog.txt)
[robotControllerLog1.txt](https://github.com/ftctechnh/ftc_app/files/1464951/robotControllerLog1.txt)
[robotControllerLog2.txt](https://github.com/ftctechnh/ftc_app/files/1464952/robotControllerLog2.txt)
[robotControllerLog3.txt](https://github.com/ftctechnh/ftc_app/files/1464953/robotControllerLog3.txt)
[robotControllerLog4.txt](https://github.com/ftctechnh/ftc_app/files/1464954/robotControllerLog4.txt)
[driverStationLog.txt](https://github.com/ftctechnh/ftc_app/files/1464955/driverStationLog.txt)

wxtrav, 2017-11-13:
Further test results: the navX seems to be an independent problem. We posted on the navX problem about that (it died multiple times without this emergency stop error ever occurring this evening).

NoahAndrews, 2017-11-13:
I saw this issue again this week, this time on a robot with a single Expansion Hub. I unfortunately don't have the logs.

gearsincorg, 2017-11-14:
@wxtrav 

> (2) one of our other teams has an issue where the two control hubs won't talk to each other at all. They are each recognized individually, but we swapped out multiple cables and it's like the port is just bad.

One possible reason for this is that they may both have the same "address".  Units come from REV with address 2.  You will need to change one to be address 3.  
Use the Advanced Settings menu on the Robot Controller app.

wxtrav, 2017-11-14:
@gearsincorg They definitely have different addresses. I've followed up with REV directly on this section and they think it's a hardware issue (possible bad port) so we'll be testing it out today.

Could you re-open this thread, concerning the Emergency Stop ("Error starting robot: java.lang.NullPointerException: Attempt to invoke virtual method 'byte[] com.qualcomm.hardware.lynx.commands.LynxDatagram.toByteArray()' on a null object reference"), since it seems like this is a big issue for a lot of teams?  Or should I start a new thread? Thanks!

I'm printing out your instructions for power-up/down and given them to the teams to paste on their robots, so we'll see if that helps.

rachmo, 2017-11-14:
@gearsincorg Regarding the two hubs not talking, we've been separately trying to help debug this from the hardware side with @wxtrav.  There is a possibility of a bad RS485 port so the recommended additional trouble shooting steps were:

1. Connect to the hubs each independently and confirm different addresses (via menu or blink codes)
2. Connect the hub together using RS485 and use the address change option under advanced settings to see if the second hub is available. A child on the RS485 bus will blink yellow under firmware ver 1_06_00 when this menu is entered. (This approach seems to have a higher success in bypassing any other software/configuration problems which can sometimes falsely report missing hub.) 
3. Repeat step 2 using the second RS485 port connection
4. Use another (3rd hub) to see if the problem can be isolated to one of the two current hubs having trouble connecting

Additionally, I can confirm that my teams have also seen the power up sequence @gearsincorg recommends does avoid communication start-up errors and that the swipe to close, and power up connect clean generally resets this kind of error.