ftctechnh/ftc_app #413: Driver station jumped back to Init during TeleOps run

FTC9915, 2017-10-16:
This behavior has been reported by many other teams. We experienced this too. It seemed happening quite frequently. I wanted to raise the awareness. Don't want people to start panicking and remind them of the initial launch of SDK issues.

People will complain this more when they are in testing after the initial build.

https://ftcforum.usfirst.org/forum/ftc-technology/52800-sdk-3-4-and-driver-station-sudden-stop

cdbbnnyCode, 2017-10-17:
Our team is having this problem also. It seems to only happen when our REV hub is attached and `hardwareMap` devices have been initialized.

ftctechnh, 2017-10-31:
Hi Folks - we have been working with the community to track issues with random stops of the Op Mode.  The behavior is that the op mode will switch to the Stop Op Mode (the default safety op mode) randomly during an autonomous or teleop run.

The Tech Team has identified two causes for this behavior.  Fixes have been introduced in v3.5 of the software to prevent random disconnects caused by these two issues. 

Details on the V3.5 changes can be found here:

https://github.com/ftctechnh/ftc_app#release-information

Instructions on how to update your software can be found here:

https://www.firstinspires.org/sites/default/files/uploads/resource_library/ftc/updating-ftc-software-instructions.pdf

gearsincorg, 2017-12-15:
This issue has been corrected in Rev 3.5 of the SDK.
It was due to the DS reacting too agressively to occasional loss of incoming telemetry.
Closing.

closed 2017-12-15 by gearsincorg