ftctechnh/ftc_app #414: Cloning TeamCode Folder

adryavitia, 2017-10-17:
I couln´t clone the TeamCoed folder and my guess is that it was because I didn´t accomplish step 2 from the ReadMe file because I can´t find the TeamCode.iml in the folder I cloned

magneticflux-, 2017-10-17:
@adryavitia There should only be `*.gradle` files when you clone the repository. The `*.iml` files are created by Android Studio and are based on the `*.gradle` files. The `*.iml` files should not be present in any repository because they are completely derived from the `*.gradle` and user settings. To get the `*.iml` files, you must import the entire project using Android Studio.

adryavitia, 2017-10-17:
And if we are using Github what would be the procedure, first I clone the repository, then I copy the TeamCode Folder and finally open Android Studio by importing the project from GitHub or from the computes

magneticflux-, 2017-10-17:
@adryavitia You should not have to copy the TeamCode folder. the rest of the repository is required for the TeamCode folder to work properly.

magneticflux-, 2017-10-17:
The steps:
1. Download the repository (Clone, zip, etc.)
2. Open Android Studio
3. Import the ftc_app project
4. Put your code in the TeamCode module

adryavitia, 2017-10-17:
Yes, but my question is what would the steps be if I want to have multiple Team Codes by cloning the main Team Code folder

fractal13, 2017-10-17:
Our team does this:
1- Copy all of TeamCode, renaming (for example TaterCode).
2- Edit settings.gradle, adding two lines:
include ':TaterCode'
project(':TaterCode').projectDir = new File('TaterCode')

If you want the new team code area to be under a different folder tree, you can use a relative path in the new File().

adryavitia, 2017-10-17:
And how do you apply the changes in the github repository?

NoahAndrews, 2017-10-17:
Same way you apply any change to the GitHub repository. Commit and push.

magneticflux-, 2017-10-17:
@adryavitia You must have your own GitHub repository though. You'll just get `push rejected` if you try to change the official FTC one. I would recommend downloading a zip of `ftc_app` and making your own GitHub repository instead of forking `ftc_app`.

adryavitia, 2017-10-17:
Thank you everyone

closed 2017-10-17 by adryavitia