ftctechnh/ftc_app #422: Warning: Expansion Hub "Expansion Hub 1" is missing

beeppatrol, 2017-10-23:
Every time I restart our robot through the robot controller I get this error:

Expansion Hub "Expansion Hub 1" is missing

I can't run any of our servos or motors.  At the moment I have tried the following:

Resetting all electronics involved several times
Changing the address of the Expansion Hub
Making new configurations(Before and after changing the expansion hub address)
Unplugging and plugging everything in again
I've made sure the phones are fully charged

Does anyone know of things I might be able to do to try to fix it?  Any help is greatly appreciated

NoahAndrews, 2017-10-23:
Try [turning off instant run](https://github.com/ftctechnh/ftc_app/wiki/Disabling-Android-Studio-Instant-Run)

beeppatrol, 2017-10-23:
What will that do?

NoahAndrews, 2017-10-23:
On some phones, if you compile the robot controller app with Instant Run enabled, the app may not be able to detect OpModes or hardware.

magneticflux-, 2017-10-23:
@beeppatrol If you ran the app using Instant Run, certain libraries may not be loaded correctly. Android Studio does some magic to reload the changed classes, but it can break complex code such as in the app.

archishou, 2017-10-23:
I don't see why it is so difficult for android studio to properly detect changes in an app, and just reload those files. I mean isn't that git?

NoahAndrews, 2017-10-23:
That's the idea, but this is not at all your typical android app. It's doing all sorts of fairly unusual stuff, and Instant Run just doesn't play nicely with it right now. 

---- On Mon, 23 Oct 2017 19:58:07 -0400 notifications@github.com wrote ----

I don't see why it is so difficult for android studio to properly detect changes in an app, and just reload those files. I mean isn't that git?

magneticflux-, 2017-10-24:
@archishou If you want an exact answer, I suggest you research the [Android Runtime](https://en.wikipedia.org/wiki/Android_Runtime), class loading, the [Dexing process](https://source.android.com/devices/tech/dalvik/dex-format), what ProGuard can do, and finally [what Instant Run actually does](https://medium.com/google-developers/instant-run-how-does-it-work-294a1633367f). It's significantly more complex than what you appear to think.

beeppatrol, 2017-10-28:
Disabling Instant Run didn't work

NoahAndrews, 2017-10-29:
Try completely uninstalling the app from the phone and then reinstalling, making sure Instant Run is disabled.

NoahAndrews, 2017-10-29:
Oh, make sure to set the hub's address back to what your configuration file expects it to be. I just noticed that you said you changed it.

ftctechnh, 2017-10-31:
@beeppatrol 
Hi beeppatrol,

I believe the Robot Controller is complaining because you changed the address of your Expansion Hub, but the configuration file references the old address.  

If you create a new config file with the new serial address and activate it, the message should disappear.  

Tom

20avva, 2018-01-04:
So if I didn't change the address of the Expansion Hub, then simply disabling Instant Run should work correct? (I'm currently at home otherwise I'd test it and find out myself :P)