ftctechnh/ftc_app #429: Large apk resulting in slow code updates

shelbyRobots, 2017-10-27:
The addition of onBot Java appears to incur a large overhead - significantly increasing the size of the generated robot controller apk.
This causes code update pushes to take MUCH longer than last year.
The apk size is nearly triple.
This seems to be an "unfair" penalty to impose upon users that are not using onBot Java.

Please find a way to allow configuration of the build to make inclusion of onBot libraries/overhead optional.

WardBenjamin, 2017-10-27:
If you're looking for a faster deploy time, you might want to check out [OpenFTC-app-turbo](https://github.com/OpenFTC/OpenFTC-app-turbo), which removes all of the bloatware used for OnBot Java and Blocks.

ftctechnh, 2017-10-31:
@shelbyRobots 
Hi shelbyRobots - we just released v3.5 of the FTC_SDK software.  It includes fixes to prevent random Op Mode stops.  It also has been trimmed down to produce a smaller APK file in order to reduce the required install time.

@WardBenjamin 
Hi WardBenjamin,
I would not characterize the OnBot Java or Blocks tools as "bloatware."  Many teams are not able to install Android Studio onto a development laptop and use it to program their competition bots.  The  OnBot Java and Blocks tools allow teams to use a light device (such as a Google Chromebook) to get started programming quickly.  OnBot Java provides the same power, speed, and flexibility as the Android Studio ftc_app project does, without the need to install and configure Android Studio.  Plus, the op modes build within a matter of seconds, without the need of the ADB utility.

WardBenjamin, 2017-10-31:
@ftctechnh while I think that OnBot Java and Blocks are good options to have, I would certainly call at least the portion of the code trimmed out in v3.5 "bloat" since it was unnecessary for even those teams using those systems. For those teams who don't use either system, and simply want faster deploy times, they might want to look into another option that strips those systems from the code. I do apologize for calling it that, you're right that those systems are great for many teams.

Also, if you haven't had a chance, I'm sure that OpenFTC would appreciate your input in https://github.com/OpenFTC/OpenFTC-app-turbo/issues/3 and connected issues/PRs.

haithcoc, 2017-11-01:
Release 3.5 improves the transfer/install time by reducing the apk size by nearly half of the 3.4 size.
It is still more than double the 2.4 size, however.  Here are the sizes for the default (non-teamcode) RC apk:
2.4 ... 8.7 MB
3.4 ... 33 MB
3.5 ... 18.7 MB
@ftctechnh 
I am willing to close this issue, but still wonder if there is some way to optionally filter apk content for teams that are not using blocks or onBot.  While I admire the initiative of the OpenFTC effort, we do not currently have the time to investigate it - nor are we in a position to take on the risk of an un-official version.
Thank you all for your hard work.

JBYoshi, 2017-11-02:
My idea: The Blocks and OnBot Java code can be restricted to the Google Play version by using `releaseCompile` in `build.gradle` for them instead of `compile`. (Although OnBot Java would need to be split out into a separate library file.) Can the references to that code be restricted to the release version as well?

ftctechnh, 2017-11-07:
Hi Folks,

Thanks for the suggestions and feedback.  We will continue to look at strategies to slip down the APK size for future releases of the software.

@haithcoc 
Let's continue to keep this issue open for now.

Tom

magneticflux-, 2017-11-11:
At [Modular FTC](https://modular-ftc.github.io/), we have two distributions for each version. The regular version contains OnBotJava and Blocks and is 19MB and the "lite" version only has the traditional OpMode support and weighs in at 8MB.

We've tried to modify the libraries at little as possible to create the lite versions, and we use completely unmodified versions of the `hardware`, `wireless-p2p`, `analytics`, and `inspection` libraries.

@JBYoshi I personally attempted to create multiple build variants for "normal" and "lite", but Maven dependencies aren't well suited to Android libraries with multiple variants. My solution is denoting "lite" versions with the "lite" suffix. You could create build variants of the final consumer of the Modular FTC libraries (i.e. the team's code), but there would be hardly any point then.

JBYoshi, 2017-12-19:
A few other ideas I thought of that could help:

- Extract all the Blocks and OnBot Java code into their own module that isn't included in the Android Studio version of the app.
- Fix Instant Run, but disable hot and warm swaps and only allow cold swaps (changes from hot and warm swaps are lost when the app restarts). The latter part can be done simply by changing the `android.optional.compilation` Gradle property (a comma-separated list of values from [this enum](https://github.com/JetBrains/adt-tools-base/blob/master/build-system/builder-model/src/main/java/com/android/builder/model/OptionalCompilationStep.java)) at runtime.