ftctechnh/ftc_app #436: Issues

ftcteam13362, 2017-11-02:
We are having some programming difficulties. When we start our program we get an error called "motorTest is not defined." We were wondering if anyone could help us solve this issue.

ftcteam13362, 2017-11-02:
We've got another problem. We also can't get the motors to turn. Everything is working fine

cmacfarl, 2017-11-02:
This is better asked in the ftc forums, or on reddit.  You are more likely to get quicker help over there.

https://www.reddit.com/r/FTC
https://ftcforum.usfirst.org/

That said, a screenshot of the error on your phone, and a copy of your opmode would be helpful.

gearsincorg, 2017-11-03:
ftcteam13362:

It would be more helpful if you entered the FULL text of any error messages.
but I assume from your error that you have not added a motor called
"motorTest" to your robot configuration (on the phone).

Also, statements like:  "We also can't get the motors to turn. Everything
is working fine" are no help at all to people who would like to assist you.
In fact, one part of that sentence MUST be wrong, because "Motors not
working" is not "fine".

When you post on the Technical forum (the better place to get help) please
provide clear details. The more information you provide, the faster we can
help you.
Also, please re-read your posts before you hit send.  Make sure your
details are correct.

​

ftcteam13362, 2017-11-07:
I'm sorry we are a 7th and 8th grade team just looking for some professional help...

gearsincorg, 2017-11-10:
And we really want to hep, but you need to give details so we can quickly predict what the problem is, and recommend a solution.

Keep at it.  Use the FTC Technology forum, and provide as much detail as you can each time.

ftcforum.usfirst.org/forumdisplay.php?156-FTC-Technology