ftctechnh/ftc_app #458: I2C connection problems on Rev Expansion Hub

JorgeNava, 2017-11-17:
# Hi !!
 Well, recently I've been having problems with the configuration file when I try to declare de Color and Gyro Modern Robotics Sensor. I'm using a code I wrote down on Android Studio the last season and it works pretty well with the Modern  Robotics controllers, but when I try to run a sensor tester code after declaring "ColorSensor" on the configuration file on 
**I2C Bus**  0,  port 0, MR Color Sensor on the 
**REV Expansion Hub**  the next message cones out: "Warning problem with : ColorSensor."
Can someone help me out with this problem? I'll appreciate it a lot, thanks.

cmacfarl, 2017-11-17:
In the other thread by forums and reddit I meant

https://ftcforum.usfirst.org/
https://www.reddit.com/r/FTC/

That said, there's a known problem with i2c and 1.6 expansion hub firmware.  Please update your firmware.  Instructions here. 

http://www.revrobotics.com/REV-31-1153/

ftc8424, 2017-11-17:
Bus 0, Port 0 is the built-in IMU.  Try making it Bus 0, Port 1 or
moving it to any of the other 3 I2C ports (they each are separate I2C
buses).

JorgeNava, 2017-11-17:
@cmacfarl Thanks I already posted it on both forums, I just wanted to recieve all the help I could. Is it wrong to post this kind of things on github? Is GitHub only for development or I can also post my doubts? (Im asking because Im learning I havent used GitHub as much as others). Thanks

JorgeNava, 2017-11-17:
@ftc8424 so when I plugg the sensor on port 3 it means that I have to configure it on Bus 3, Port 3? Or on Bus 3, Port 1?

ftc8424, 2017-11-17:
Bus 3, Port 0

gorpong, 2017-11-17:
When you're in the Edit Configuration stage you can select I2C Bus 0 and click "Add" to add another port on it, that would make it Port 1 on Bus 0.  My team's done that with our color sensor and it works fine on the same bus as the IMU.

When you move it to another bus, I think you'll need to do the "Add" to add Port 0 and then configure the type of sensor and give it the name.

You can try both ways and see which works best for you.  My team doesn't seem to have any problems with the REV color sensor sharing the same bus as the built-in IMU.

rachmo, 2017-11-17:
Do you have a level shifter and sensor adapter cable for your MR Color Sensor? Are they plugged in with the correct orientation?

[image]

cmacfarl, 2017-11-17:
@ftc8424 The idea of a port in the context of a device on an i2c bus is a misnomer.  An i2c bus intrinsically has no notion of ports.  All devices are uniquely identified on a common bus by their i2c address.  As long as the color sensor's address is different than the IMU's address this should work properly.

@JorgeNava Yes.  Questions asking for general help are better off in the forums.  The github issue tracker should ideally be reserved for bugs found in the SDK.

ftc8424, 2017-11-17:
That makes sense, but I know that when I had the configuration file
where I defined the port 0 to be "color sensor" on bus 0, it always gave
me an error.  I had to go into bus 0 and click "Add" to add a new port,
and set that port to be "color sensor" and make port 0 "imu" in order to
make them both work.  So, port in the configuration file *is* important
and relevant, even if it doesn't make sense in a bus/i2c concept.

cmacfarl, 2017-11-17:
Hmm, that sounds like a legitimate bug.  I would think if you aren't going to use the IMU, you should be able to replace it's definition in the configuration with an i2c device on bus 0 that you are going to use.  And in fact if this restriction were firmly in place, then the user interface should not allow it at all.  There's no harm in not doing so, but if you can't, and there's no underlying technical reason why, I can see how these confusions may arise.

JorgeNava, 2017-11-17:
@rachmo yes they are. 
Thanks @gorpong and @ftc8424 I'll give it a try today and notify you about what happened if you dont matter.

closed 2017-12-03 by cmacfarl

JorgeNava, 2017-12-03:
Thanks @cmacfark for your help, it was the I2C address if anyone wanted to know.