ftctechnh/ftc_app #469: REV Expansion Hub and "problem with "MRSensor""

technova12611, 2017-11-28:
Our team had seen similar issues as described in the FTC Forum thread below. Although it didn't seem to lock up the thread anymore in REV firmware 1.7, sometimes the MR sensor (Range sensor for us) stopped reading values. Also, we had seen one "problem with" issue on the embedded IMU, which made the autonomous program off track. These i2c issues seemed happening to many teams, there are postings in reddit forum too.

This issue seems highly reproducible, simply add 2-3 MR range or color sensors and using the sample SensorMRRangeSensor program to getDistance() on each one of them, you will get "problem with" issue very often. It seemed that the sensor could recover from this, but sometimes it didn't. This issue definitely hinders team's ability to create more advanced autonomous programs.

https://ftcforum.usfirst.org/forum/ftc-technology/57876-rev-hub-and-mr-i2c-problem-with-issue

https://www.reddit.com/r/FTC/comments/7g1xaz/warning_problem_with_imu/

JorgeNava, 2017-12-03:
Did you try changing the I2C address to the default address?

aedancullen, 2018-03-20:
We have seen this many times in practice. It's definitely common; it doesn't really seem to happen often enough to be a major roadblock though (it's never happened in competition...) This isn't because a sensor isn't plugged in, or because a sensor is misconfigured. The internal IMU and/or our color sensors will sometimes produce this message in the middle of running an opmode (it'll be reading data fine until a point at which the message appears mid-autonomous-run.)
 An interesting thing is that power-cycling the REV modules and allowing ftc_app to rediscover them does not fix the problem - ftc_app always needs to be restarted (at least on our robot with ftc_app 3.6 and REV firmware 1.7.2) in order for it to clear the "Problem with <some I2C device>" message. We haven't modified any I2C addresses, and each I2C device is alone on its own bus.

I suspect it has something to do with power. Upon running the robot for a long time with a not-fully-charged 12V battery (in practice sessions, for example) seems more likely to reproduce it than a competition environment when we're running single matches, with fresh batteries in-between. And, incidentally, it always seems to happen to us after the point during our autonomous when the drivetrain is accelerating extremely fast, drawing a lot of current very quickly. A lot of I2C devices require initialization writes to happen before data can be read (data format settings and such) so if there's noise or voltage transients somehow ending up on the 3.3V rail then maybe the sensor is in an uncertain state where I2C operations are going to time out. Possibly @robogreg might know something about how the 3.3V is supplied internally (is it an LDO or DC/DC buck converter off of the 12V?)  It seems like this problem appears more often with external sensors than with the internal IMU though (and maybe even more often when you've got a level shifter running its 5V boost converter on top of the 3.3V supply). Looks like we need to try adding some extra bypass caps and see if that helps.

Azuresword, 2018-07-17:
we have deal with this problem,when you face the "problem"problem,first of all,check your firmware,make sure it is upload to the newest edition.then make sure the most higher load REV controller hub not connected to the I2C sensor. I am sure it will solve your problem.

slylockfox, 2019-02-13:
At a qualifier in Missouri last weekend, a team told me they were having trouble using two MR range sensors, and I think it was this same symptom.  Rather than connecting the sensors to two separate busses on the same REV hub, they said they had the sensors connected to two different hubs, and still had the issue. Any new ideas on this issue?  (@rvanhoose the team was Rolla Patriots)

Xbots, 2019-02-13:
Our experience in using 2 MR range sensors with Rev hubs is similar,
especially with "problem" problem showing up intermittently and programmers
having to anticipate it all in the code. Changing the level shifter fixed
it a couple of times. Prior to using MR range sensors we actually used REV
distance sensors and while they don't need level shifters and hence less
reasons to fail, they seemed to have worked more reliably while we were
still working out other bugs, so we might switch back.

Have you used Rev distance sensors instead of MR sensors? What issues did
you see with those sensors?

Thanks

slylockfox, 2019-02-13:
The team said the REV distance sensors gave inaccurate numbers, so they preferred the MR sensors, but I think they might have been referring to the old REV color/distance sensor, and not the new LIDAR sensor.  I'm still trying to get followup communication with the team