ftctechnh/ftc_app #474: Is there any way to access "isOpModeActive()" from outside of a class that extends LinearOpMode

LBYPatrick, 2017-12-03:
I'm trying to write a support class for our autonomous mode since the four different autonomous modes for different spots actually looks similar. I really want it to separate from registered OpModes. Does anyone have ideas about this? Thx.

cmacfarl, 2017-12-03:
This is better asked on the FTC forums or on r/FTC on reddit.  

But to answer your question, pass a reference to your linear opmode into your support class.  e.g. From the class you've derived from LinearOpMode.

closed 2017-12-03 by cmacfarl

qwertychouskie, 2017-12-03:
@LBYPatrick Alternatively you could pull `FtcMenu`, `FtcChoiceMenu`, `FtcValueMenu`, `HalDashboard`, `HalDbgLog`, `TrcDebugTrace`, and `TrcUtil` from <https://github.com/trc492/FtcSamples>, see https://github.com/FTCTeam10298/2017-18-code/tree/master/TeamCode/src/main/java and specifically https://github.com/FTCTeam10298/2017-18-code/blob/master/TeamCode/src/main/java/org/firstinspires/ftc/teamcode/OurM0Bot_Autonomous.java for examples.

LBYPatrick, 2017-12-04:
@cmacfarl Thanks for the info. 

So for ``LinearOpMode opmode;`` do I need to declare it as ``static`` for receiving the reference? (I had an issue on passing as reference with DcMotor class from the FTC Official) Such as:

In SupportClass.java
```

final public class SupportClass {

       private static LinearOpmode opModeObj; // The thing that I am talking about 

       public SupportClass(LinearOpMode opmode) {opModeObj = opmode;}
       public SupportClass() {};

       public boolean reportStatus() {
             return opModeObj.opModeIsActive();
      }
}

```

And in my LinearOpMode: 
```

final public someOpMode extends LinearOpMode {

@Override
public void runOpMode() {
              SupportClass auton = new SupportClass(this);
              telemetry.addData("OpMode Status: ", auton.reportStatus()));   
     }
}

```

Does this work?

calebsander, 2017-12-04:
No need for `static`. `static` would imply that `opModeObj` is a property of the universe of *all* `SupportClass` objects, rather than a property of a *single* `SupportClass` object. You should be able to simply omit the keyword `static` to make it an instance property. You may also want to remove the constructor that takes in no parameters, as this suggests that `SupportClass` can be used without a `LinearOpMode`. This will likely lead to NPEs when you call methods like `reportStatus()`.

Also, you are missing a `class` keyword in the declaration of the `someOpMode` class.

In addition, the `final` keyword is a rare modifier to see on a class. Adding `final` just means that you can't subclass the class, which is generally useful only for very specific circumstances.

LBYPatrick, 2017-12-04:
@calebsander muchos gracias! I will try it on our robot(s) today.