ftctechnh/ftc_app #477: Can we have a method to temporarily disable the "stuck in loop" watchdog?

aedancullen, 2017-12-06:
This is a feature request.

Could we add a method to temporarily disable the "stuck in loop" watchdog in ftc_app? Our team uses a custom autonomous navigation library, and it'd be really nice to be able to temporarily jump into a sort of "linear" programming style within one iteration of loop() with the watchdog disabled and then reenable it when that routine is done (and when the program might go back to a continuously-looping state). There are undoubtedly many applications in which this type of 'hybrid' capability would benefit (saves the effort of writing stateful code...) and all that would be required would be 'enableWatchdog' and 'disableWatchdog' methods, for example (if there aren't internal USB/network comm. routines that depend on the occurrence of loop() method returns, that is...)

Obviously, nobody should be disabling the watchdog to prevent 'stuck' problems that are the results of a real programming mistake. But allowing the watchdog to be disabled for programmers who know that they are taking more time in loop() on purpose will be very helpful for those who want opportunities to use a more free-form autonomous programming style (and write cleaner code, more maintainable code, etc.)

rgatkinson, 2017-12-06:
In the constructor of your opmode, you can change the timeout value used. I'd have to look up the member variable name (not in front of me), but it's pretty obvious, if I recall.

aedancullen, 2017-12-06:
Oh, that's cool. Thanks, that's really the solution.

closed 2017-12-06 by aedancullen

cmacfarl, 2017-12-07:
For documentation purposes...

The timeout values are declared public in the OpMode class.   My team had to redefine msStuckDetectInit last year because the Vuforia initialization would occasionally trip the watchdog.  As the comment says though, do be careful and only change what you absolutely need to the smallest time necessary...

```
  //----------------------------------------------------------------------------------------------
  // Safety Management
  //
  // These constants manage the duration we allow for callbacks to user code to run for before
  // such code is considered to be stuck (in an infinite loop, or wherever) and consequently
  // the robot controller application is restarted. They SHOULD NOT be modified except as absolutely
  // necessary as poorly chosen values might inadvertently compromise safety.
  //----------------------------------------------------------------------------------------------
```

gearsincorg, 2017-12-07:
Since you are a "programmer who know that they are taking more time in
loop()", perhaps you could break the init process into a couple of smaller
steps.

You could sequence these over a couple of loop cycles.  Just setup a simple
state machine with two or more states.

This way you would still get the same init, but you wouldn't be
circumventing the safety protocol that has been put in place to ensure your
opmode responds to STOP requests within a suitable time.

Phil.
​