ftctechnh/ftc_app #484: "No Devices Found" error even when devices are found

cdbbnnyCode, 2017-12-15:
When scanning for devices, the driver station shows a 'No Devices Found' error, even if devices are found.

[image]

This is mostly a visual bug, and doesn't affect the usability.

gearsincorg, 2017-12-15:
This is one I haven't seen.  Is there anything you can tell us about how/when this error comes up.

eg:  
I notice there is no active config file.. is this a new phone setup, or had the old config been deleted?
Does this come up when you first enter the config system, or is it the result of hitting the scan button?
If you hit scan, does the same condition repeat itself?
Have you seen this more than once?
Anything else notable ?

cdbbnnyCode, 2017-12-15:
This is not a new phone setup (there are other configs available); I was creating a new config.  
I believe this is a v3 bug because I don't remember seeing anything like it last season, but I do remember having the problem with the other configurations from this season.  
It only seems to happen when the config is first created and has not been saved, and will happen even if the 'Scan' button is pressed multiple times.  
I do not know if this error happens if devices are connected when opening the configuration creator, though.  

My theory is that the code that pops up the message checks the *saved* file for devices, and if there are none (e.g. the file wasn't even saved yet), it shows the message.

gearsincorg, 2017-12-15:
More questions...

I see this is happening on the Driver Station.  Have you ever seen this while configuring using the Robot Controller directly (wondering if it's the RC and DC getting out of sync).

So, at this point, have you configured motors etc for the attached REV hub, but not saved them yet?
When does the "no devices found" message actually pop up?    eg:  As you exit the Hub setup page and return to the top most menu, or some other point in the sequence.  When is the FIRST time it shows up?  

From your comment "I do not know if this error happens if devices are connected when opening the configuration creator", it sounds like you have connected the REV hub AFTER opening the configuration utility on the Driver station.  Is this correct?  This could be the trigger...

cmacfarl, 2017-12-15:
There's definitely a state disconnect there.  That message appears if a configuration has been parsed, and software thinks it's empty.  This is a "shouldn't happen" scenario when you are creating a new configuration.  I have not seen this either, which implies there's still something unknown about the path of actions that led you to this result.    And there are a number of paths that can lead to that message being displayed.   Answers to Phil's questions would be helpful.

Thanks.

cdbbnnyCode, 2017-12-16:
I have run a few tests and found some surprising results:
* All of these happen on both the driver station and the robot controller.
* When no devices are detected initially, the software shows a message with an OK button, which will dismiss the message.
  * Dismissing the message will cause all further messages to disappear (even if no devices are found)
  * If the message is not dismissed, any later scans will show the message *without an OK button*.
* If devices are detected at the start, all further messages will disappear immediately after they are shown.

gearsincorg, 2017-12-16:
I can replicate what your are reporting in this last post, but I'm not sure
if it's related.

When I hit the scan button for a second tome, I see an additional prompt,
asking about losing unsaved changes.  I think this popup is actually
preventing you from seeing the "missing" no-devices message.

I say this because on my ZTE phone, when I click the OK button on the
Unsaved Changes dialog, I see a momentary splash of orange, which I suspect
is the no-devices popup being removed.

So it certainly looks like there is a clash of popups here, but I'm not
sure if/how that is related to the originally reported User Interface
confusion.

Phil Malone
GEARS Inc.

phil.malone@mr-phil.com
Office: 301.387.2331
Mobile: 301.501.7424