ftctechnh/ftc_app #485: How to disable some unnecessary service

berberman, 2017-12-17:
As you know, while we are using RobotControllerApp, various service that registered when the activity start is running. For example, we don't need to use "block programming" but http service still running, it cause the electricity consume badly. Please give solution for us to disable such service through a convince way.
Thanks.

magneticflux-, 2017-12-17:
@berberman You can check out the [Modular FTC](https://modular-ftc.github.io/) and [OpenFTC](https://github.com/OpenFTC) projects if you wish to customize the Robot Controller more. If you end up using them however, you should pay attention to the legality of such modifications.

NoahAndrews, 2017-12-19:
Specifically, on the OpenFTC side, [OpenFTC Turbo](https://github.com/OpenFTC/OpenFTC-app-turbo/) strips out the blocks and OnBotJava features.

JBYoshi, 2017-12-19:
Related: #429.

dmssargent, 2018-01-03:
@berberman How do you know the HTTP server running to support Blocks is causing excess battery drain versus something else?
Do you mind linking reports you may have generated about the issue?

berberman, 2018-09-18:
Not only that caused the apk very large, but it really slowed down operating and increased battery consumption as well. #429