ftctechnh/ftc_app #486: Gamepad Disconnection with DS 3.5

WardBenjamin, 2017-12-18:
We're having exactly the same issue as https://ftcforum.usfirst.org/forum/ftc-technology/59140-ftc-app-3-5-build-and-logitech-controller-behaviour

Our controllers will only rarely connect to the DS. This was not happening pre-3.5. We've tried 3 different DS phones, 3 OTG adapters, and 4 gamepads. All gamepads are switched to the correct mode and we attempt to connect with Start+A or Start+B. RC SDK version does not make a difference. This occurs on ZTE Speeds, Galaxy S5s (Verizon), and an unapproved phone.

Downgrading to DS 3.4 fixes the issue.

We have confirmed that the cable, gamepad, and phone are all working (the gamepad is usable to scroll through icons on the home screen).

WardBenjamin, 2017-12-18:
LED was lit on our USB hub. We stopped using the USB hub after the first few tries to remove a variable. Restarting the phone(s) did not fix the issue. This occurs on all of our phones that we have tested. We are using Logitech controllers.

qwertychouskie, 2017-12-18:
Maybe try completely uninstalling the DS app, then reinstalling from the Play Store (or [Yalp Store](https://f-droid.org/packages/com.github.yeriomin.yalpstore/)).

WardBenjamin, 2017-12-18:
Completely reinstalling does not fix the problem.

gearsincorg, 2017-12-19:
I've been using 3.5 since it's release with ZTEs and Moto G2 & 4 without
any more than the occasional failure to connect, and this is usually due to
the controllers being plugged in when the phones are powered up.

Just this weekend, our drive team had a drive practice (using ZTE's) and
ran 15 matches in a row with no issues.

Not saying you aren't having an issue, it's just that my experience says
it's not a 3.5 issue.

Phil Malone
GEARS Inc.

phil.malone@mr-phil.com
Office: 301.387.2331
Mobile: 301.501.7424

WardBenjamin, 2017-12-19:
It doesn't seem to be happening to most teams, but I have a hard time believing that the issue is _not_ with 3.5 since the gamepads are definitely connected to the phones (as they can be used on the home screen), and downgrading to 3.4 fixes the connection problem. With those circumstances, I don't see how it can be anything else than the DS app.

Any other suggested fixes? Obviously we can't run a mismatched DS/RC SDK version in competition.

gearsincorg, 2017-12-19:
You have presented many different hardware combinations, so it's hard to
address the specific problem.

Can you humor me with "one" specific combination of hardware that causes
the problem.

Can we focus on a ZTE speed phone, with a standard Logitech F310 gamepad
with NO hub connected, and just using a simple USBA(F)- MicroUSB(M) cable.
Ensure that the DS app is Rev 3.5.

Assuming that you startup the phone from a power down (with the Gamepads
disconnected at the Micro USB plug)
1) Power on the phone.
2) Start the DS app, verify that on the settings page, a Logitech F310
Gamepad is selected.
3) Plug in the Gamepad.
4) Press Start +A

Nothing?

Phil Malone
GEARS Inc.

phil.malone@mr-phil.com
Office: 301.387.2331
Mobile: 301.501.7424