ftctechnh/ftc_app #492: Teleop Uninitializes by itsself

kuriosity-robotics, 2018-01-02:
We have a problem with our teleop because it randomly uninitialize by itself without us touching the screen. We also changed phones and restarted them and it still does that. If anyone could help that would be extremely helpful.

NoahAndrews, 2018-01-03:
What version of the RC and DS apps are you using? It sounds like your DS in particular may not be running 3.5.

kuriosity-robotics, 2018-01-03:
we are using version 3.5

NoahAndrews, 2018-01-03:
Double check the about screens? I've seen teams think they've updated before only to be mistaken.

JBYoshi, 2018-01-04:
Maybe try 3.6? It was released a few weeks ago.

Stiffy2000, 2018-01-19:
Is there a ```waitForStart();``` line before the ```while (opModeIsActive()) {``` line?

kuriosity-robotics, 2018-01-20:
Yes there is

bqrichards, 2018-02-09:
Hey @kuriosity-robotics! Do you mind posting your TeleOp so we could inspect the code?