ftctechnh/ftc_app #500: Build FAILED!

mikemazp, 2018-01-09:
when building,only had the following:

Build started at Tue Jan 09 2018 13:43:08 GMT+0800 (中国标准时间)

Build finished in 0.7 seconds
Build FAILED!

Can I find more info ...

NoahAndrews, 2018-01-09:
Look in the gradle output tab (bottom left)

mikemazp, 2018-01-11:
[image]
[image]
I couldn't find anything...

mikemazp, 2018-01-11:
Any java file building,always build fails.
Why...

NoahAndrews, 2018-01-11:
Sorry, I assumed you were using Android Studio.

NoahAndrews, 2018-01-11:
I'm not sure how much logging OnBotJava does, but can you post your log file?

Stiffy2000, 2018-01-12:
Were you connected to the controller when you were building? 
When this happens to me, I just repeatedly click the build button.

mikemazp, 2018-01-15:
@Stiffy2000 ,I connected to the controller when building.And I tried more times;
@NoahAndrews  I weren't using IDEA.

mikemazp, 2018-01-15:
I succeed to building only once ，and synchornized  the controller code；then no more success。

mikemazp, 2018-01-15:
So it made me feeling so strange..

cmacfarl, 2018-01-15:
Can you please pull the logs off of the robot controller?  Long click on
the logs on the View Logs activity and you can email them to yourself (if
your phone is connected to the wider internet).

Stiffy2000, 2018-01-16:
Copy the code and save to your computer.
Then, reinstall
See if that works.

BotsBurgh

mikemazp, 2018-01-19:
@Stiffy2000 I will try it.