ftctechnh/ftc_app #502: Random Op Mode Stoppage Bug

NickDim, 2018-01-10:
I have noticed a bug where opmodes get randomly stopped. This bug can occur at any point in the opmode and happens without giving an error or any message, the driver phone acts as if nothing happened. My team has found that this can mostly be avoided by restarting the app.

NoahAndrews, 2018-01-10:
What version of the apps are you using?

NickDim, 2018-01-10:
v3.5

NoahAndrews, 2018-01-10:
Can you post a robot controller log file that includes a random OpMode stop?

aedancullen, 2018-01-12:
I think 3.6 has a fix for this.

NoahAndrews, 2018-01-12:
It's 3.5 that has that fix.

gearsincorg, 2018-01-12:
BTW, The Random stopping bug was a problem caused by the Driver station
App, so it's most important to ensure that the DS has version 3.5

Sometimes teams forget that this phone also needs to be updated.

From your initial comment you say that " the driver phone acts as if
nothing happened".

This does not sound like the original "Stopping opmode" bug.
In that case the driver station phone actually shows a "stopped" robot.

Can you be more specific as to what you are seeing?
Do you mean that the robot "shows that it is running", but it does not
respond to driver input?
This could be a disconnected driver station.....
If this is the case you would not see any activity on the two gamepad icons
on the top of the DS phone.
By clicking Start A could restore control.

Phil.

Phil Malone
GEARS Inc.

phil.malone@mr-phil.com
Office: 301.387.2331
Mobile: 301.501.7424

gearsincorg, 2018-01-18:
@NickDim  Can you verify that your Driver Stationd is ALSO rev 3.5 or 3.6  (It's the pre 3.5 driver station that causes the random stopping bug)

richiksc, 2018-02-09:
@gearsincorg It is