ftctechnh/ftc_app #503: Continuous blinking light during scan

emitt411, 2018-01-11:
My team recently got a REV Expansion Hub. When we plug it in and scan for the portal, it does not appear. What also happens is that the light on the hub will turn green, the continue to blink blue without stopping. Everything is plugged in correctly based on the expansion hub guide. Please help as soon as possible.

Stiffy2000, 2018-01-12:
1. When you plug it in without the battery attached, does it [The REV hub] turn on? If not, check your USB cable. It's probably a problem with that. 
1. [Read the REV Hub Documentation](http://www.revrobotics.com/content/docs/REV-31-1153-GS.pdf)
  * The LED blink codes are on page 17. 
1. Have you reinstalled the Driver station app? The robot controller app?

emitt411, 2018-01-12:
While plugged in to the phones or a computer, it receives power. The when we scan, the light turns green then blinks blue, but doesn't stop blinking. Next time I get the chance, I will reinstall the apps. If there is anything else you would like to know, just ask.

gearsincorg, 2018-01-12:
The LED is meant to do a LONG green flash,and then blink the Module address
with blue blinks.

If it never stops blinking, perhaps the address has been set very high.

Phil Malone
GEARS Inc.

phil.malone@mr-phil.com
Office: 301.387.2331
Mobile: 301.501.7424

emitt411, 2018-01-12:
I've set the address to 2 yet it blinks constantly without the long green after it. It doesn't show the portal at all, and when I scan again, the green light comes back on, but then continues to blink blue with no sign of stopping.

cmacfarl, 2018-01-12:
> Office: 301.387.2331 <(301)%20387-2331>
> Mobile: 301.501.7424 <(301)%20501-7424>

Are you using Android Studio?  If so is instant run on?

jdkunzeman, 2018-01-12:
According the Rev Expansion Hub documentation, the blinking blue light means "Keep alive has timed out. Fault will clear when communication resumes."

NoahAndrews, 2018-01-12:
Can you provide the Robot Controller log file?

emitt411, 2018-01-12:
I will get log file tomorrow when I can.

aedancullen, 2018-01-12:
Is the hub connected to 12V power? If not, try it with the 12V battery connected (possibly increased load on the phone and hub microcontroller is causing the 5V USB OTG rail to droop for some reason and the microcontroller is resetting (restarting its firmware) during the scan due to a brown-out? And, what phone are you using? At least on ZTE Speed, we've sometimes seen weird things where when the phone CPU load is high, and the REV is powered only by USB, the LED on the REV flickers (maybe due to the 5V rail drooping). So this could be a similar thing.

emitt411, 2018-01-12:
We always connect it to a 12V battery before turning it on. We are currently using Moto G's for it, but haven't tried or ZTE phones yet. The phones are fully charged before use as well. I will get a picture of the log file later on.

NoahAndrews, 2018-01-12:
Make sure you provide the full (very long) log, not just a screenshot.

cmacfarl, 2018-01-12:
It sends the entire log.

emitt411, 2018-01-14:
I can't seem to copy the log. It says it's an empty file after I long press and try to send it to my email.

cmacfarl, 2018-01-15:
The log is sent as an attachment.  I just verified on my phone that this works.  Can you please provide some screenshots and perhaps the steps that you took to reproduce this?  Here's a screen capture of gmail showing receipt of the log I just sent...

[image]

rsshilli, 2020-09-20:
We had this problem. I had my WIFI channel set to channel 2. I changed it to auto, unplugged my phone and the battery from the robot, plugged in the battery first, then the phone, and then it worked. I'm surprised the WIFI channel would matter, but I figured I'd share for the next person.