ftctechnh/ftc_app #505: Gyro sensor turn twice

maylee0515, 2018-01-12:
I want to make a code that makes my robot turn 90 degrees to the right and then turn 180 degrees so that the robot is facing 90 degrees left from the initial position. I can't figure out what I need to put in or what I need to change in order for the robot to do what I want it to do. Currently, at the moment, it starts turning 90 degrees right and keeps turning in circles. This is my code so far...
`@Override
    public void runOpMode() throws InterruptedException{
        leftfront = hardwareMap.dcMotor.get("leftfront");
        leftback = hardwareMap.dcMotor.get("leftback");
        rightfront = hardwareMap.dcMotor.get("rightfront");
        rightfront.setDirection(DcMotor.Direction.REVERSE);
        rightback = hardwareMap.dcMotor.get("rightback");
        rightback.setDirection(DcMotor.Direction.REVERSE);
        gyro = hardwareMap.gyroSensor.get("gyro");
        gyro = (ModernRoboticsI2cGyro) gyro;
        gyro.calibrate();
        touchsensor = hardwareMap.touchSensor.get("touchsensor");

        waitForStart();
        leftfront.setPower(basestart);
        leftback.setPower(basestart);
        rightfront.setPower(basestart);
        rightback.setPower(basestart);
        sleep(2500);
        rightfront.setPower(basepower);
        rightback.setPower(basepower);
        leftfront.setPower(0);
        leftback.setPower(0);
        while(opModeIsActive()){
            while(gyro.getHeading() == 270) {
                rightback.setPower(0);
                rightfront.setPower(0);
            }
            sleep(1000);
            leftfront.setPower(basepower);
            leftback.setPower(basepower);
            while(gyro.getHeading() == 90){
                leftback.setPower(0);
                leftfront.setPower(0);
            }
                sleep(1000);`

NoahAndrews, 2018-01-12:
You need to allow some amount of error in your target heading. If the gyro never returns values of precisely 270 and 90, it will never stop turning. 

Also, you should be using encoders to go forward a certain amount, not timing. And you you'll get more precise turns if you run both motors at the same speed but in opposite directions.

cmacfarl, 2018-01-12:
Noah has set you on the right course, but this is better handled on r/FTC over on reddit, or the official FTC forums.

closed 2018-01-12 by cmacfarl