ftctechnh/ftc_app #506: Error running TeamCode: Default Activity not found

J053Fabi0, 2018-01-12:
I have this issue since the new update of Android Studio. Even with the ftc_app original project, without modifying it. My friend is having the same problem, so its not only me.

[image]

[image]

### Any idea on how to solve it?

cmacfarl, 2018-01-12:
What happens if you select File -> Invalidate Caches / Restart... ?

J053Fabi0, 2018-01-12:
I have tried that before... without luck

cmacfarl, 2018-01-13:
What are the contents of your AndroidManifest.xml file?   Located under FtcRobotController -> manifests?

J053Fabi0, 2018-01-13:
`<?xml version="1.0" encoding="utf-8"?>
<manifest xmlns:android="http://schemas.android.com/apk/res/android"
          xmlns:tools="http://schemas.android.com/tools"
          package="com.qualcomm.ftcrobotcontroller"
          android:versionCode="21"
          android:versionName="3.1">

  <uses-permission android:name="android.permission.RECEIVE_BOOT_COMPLETED" />

  <application
    android:allowBackup="true"
    android:largeHeap="true"
    android:icon="@drawable/ic_launcher"
    android:theme="@style/AppTheme" >

    

    

    
    <service
      android:name="com.qualcomm.ftccommon.FtcRobotControllerService"
      android:enabled="true" />

    

  </application>

</manifest>`

J053Fabi0, 2018-01-13:
<?xml version="1.0" encoding="utf-8"?>
<manifest xmlns:android="http://schemas.android.com/apk/res/android"
          xmlns:tools="http://schemas.android.com/tools"
          package="com.qualcomm.ftcrobotcontroller"
          android:versionCode="21"
          android:versionName="3.1">

  <uses-permission android:name="android.permission.RECEIVE_BOOT_COMPLETED" />

  <application
    android:allowBackup="true"
    android:largeHeap="true"
    android:icon="@drawable/ic_launcher"
    android:theme="@style/AppTheme" >

    

    

    
    <service
      android:name="com.qualcomm.ftccommon.FtcRobotControllerService"
      android:enabled="true" />

    

  </application>

</manifest>

J053Fabi0, 2018-01-13:
http://josefabio.me/AndroidManifest.xml

J053Fabi0, 2018-01-13:
[image]

magneticflux-, 2018-01-13:
@J053Fabi0 It should be [this](https://github.com/ftctechnh/ftc_app/blob/6370c4e556d6f6f1069659372323cfbbbc7f2dcb/FtcRobotController/src/main/AndroidManifest.xml).

J053Fabi0, 2018-01-13:
[image]

magneticflux-, 2018-01-13:
@J053Fabi0 Those references are red either because the project has never been successfully built or because the files they reference are missing. Those files should be (and are) included in the current latest version and prior versions.

cmacfarl, 2018-01-13:
You are definitely missing the activity declaration in the first manifest file you posted.  

What is your process for obtaining the FTC SDK?  If you switch to a new empty directory, do a git clone https://github.com/ftctechnh/ftc_app.git, (assuming of course you have a git client on your machine) open that project and build, do you get the same error?

J053Fabi0, 2018-01-13:
@cmacfarl The error doesn't show if the project is the original O_o
But I have this warnings:
[image]

J053Fabi0, 2018-01-13:
Nvm, I solve them. Thanks for the help guys! Ur the best

closed 2018-01-13 by cmacfarl