ftctechnh/ftc_app #507: HSR-2645CR servo problem with Rev expansion hub

jdkunzeman, 2018-01-14:
We want to use the above servo because it is more powerful.  Our robot is build with two Rev expansion hubs.  I have a simple autonomous program which runs a CR servo in one direction for three seconds, then the other direction for three seconds, then sets it to the mid-position for three seconds so that it doesn't move.  The program works with the above servo using a MR CPDM and servo controller.  It works on a Rev hub with a Tetrix CR Servo 39177,  but it doesn't work with the Rev Hub and the above servo, so far.  Via telemetry statements I can tell that the program is running, but the servo never moves.  

In the log, I see this:
01-13 20:53:23.551  9517  9607 I RobotCore: Attempting to switch to op mode ContServTest
01-13 20:53:23.741  9517  9607 W LynxPwmOutputController: LynxSetPWMEnableCommand not implemented by lynx hw; ignoring
01-13 20:53:23.751  9517  9607 W LynxPwmOutputController: LynxSetPWMConfigurationCommand not implemented by lynx hw; ignoring
01-13 20:53:23.751  9517  9607 W LynxPwmOutputController: LynxSetPWMEnableCommand not implemented by lynx hw; ignoring
01-13 20:53:23.761  9517  9607 W LynxPwmOutputController: LynxSetPWMConfigurationCommand not implemented by lynx hw; ignoring
01-13 20:53:23.761  9517  9607 W LynxPwmOutputController: LynxSetPWMEnableCommand not implemented by lynx hw; ignoring
01-13 20:53:23.771  9517  9607 W LynxPwmOutputController: LynxSetPWMConfigurationCommand not implemented by lynx hw; ignoring
01-13 20:53:23.771  9517  9607 W LynxPwmOutputController: LynxSetPWMEnableCommand not implemented by lynx hw; ignoring
01-13 20:53:23.771  9517  9607 W LynxPwmOutputController: LynxSetPWMConfigurationCommand not implemented by lynx hw; ignoring

Any idea what's going on?  Any quick fixes or workarounds for the Rev hub?  This is a digital servo, I think.  If it's not supposed to work with Rev, I must have missed where that is documented.  I am using the CRServo function.  

Here's the code:
public class ContServoTest extends LinearOpMode {
    CRServo crs;
    @Override
    public void runOpMode() {
        crs = hardwareMap.crservo.get("left_hand");
        telemetry.addData(">", "Press Start to scan Servo." );
        telemetry.update();
        waitForStart();
        while(opModeIsActive()){
            crs.setPower(-1);  //Run servo in one direction
            telemetry.addData(">", "First dir");
            telemetry.update();
            sleep (3000);
            crs.setPower(1);  //Run servo in opposite direction
            telemetry.addData(">", "Second Dir");
            telemetry.update();
            sleep (3000);
            crs.setPower(0);  //Stop servo
            telemetry.addData(">", "Middle");
            telemetry.update();
            sleep (3000);
        }
        telemetry.addData(">", "Done"); // Signal done;
        telemetry.update();
    }
}

If needed, I can try to create a log with run on MR, run different servo on Rev, and run the HSR servo back to back to back and send that.

gearsincorg, 2018-01-14:
The rev servo pulses are probably out of range for this servo. Simple
solution is to decrease  the range. Instead of sending -1 and 1 power, send
-0.75 and 0.75. This will keep the pulses within range.

jdkunzeman, 2018-01-14:
@gearsincorg, that worked!  The servo started working at 0.8.  Thanks very much!

I had been under the impression that they app would account for the differences between servos.

closed 2018-01-14 by jdkunzeman

cmacfarl, 2018-01-14:
The SDK, and/or REV firmware, has no notion of what sort of servo is plugged into the port.  The configuration just states it's a servo, and there no way for the software to query the servo and get an answer as to a model number.  And even if servo model numbers were baked into the configuration there are literally 100s of servos making this in all practicality, infeasible.  There's a similar issue with the Hitec winch servo (model number escapes me at present).

NoahAndrews, 2018-01-14:
HS785HB I think. I've spent a lot of time with that one.