ftctechnh/ftc_app #509: Vuforia front camera very dark in slightly dim lighting conditions

NiknAknO, 2018-01-19:
This issue doesn't happen in the camera app. Happened to two Samsung Galaxy s5 phones.

premkk, 2018-02-01:
We had this same issue. Looks like an issue with vuforia (see link below). Can we get a patch for this in the FTC SDK soon?

https://developer.vuforia.com/forum/android/extremely-dark-camera-andoird-device

aedancullen, 2018-02-03:
The Vuforia devs will have to fix that. As far as I know ftc_app has nothing to do with lower-level camera init/configuration - it just calls into a binary blob for the Vuforia API.

NoahAndrews, 2018-02-03:
The point is that the latest version of vuforia apparently already fixed this. FTC needs to update.

premkk, 2018-02-03:
This is right. I downloaded the vuforia jar and jni and patched it in the
FTC SDK directories in Android Studio but ran in to runtime exceptions.