ftctechnh/ftc_app #511: Phone Disconnection Randomly During Programs + Initilization

NeonOctopus, 2018-01-21:
When our phones are laying untouched, either initialized or in the middle of a program, the phones disconnect, and reconnect. The ping will rise to 500, then dip down to a low number, like 3, and then after resetting, it will explode up to about 500. We are using version 3.6 on both phones. We have Moto G4s. We used a second set of phones, and the same problem prevailed. If it helps solve the problem, we are using blockly. This is happening in autonomous and tele op.

KitKatTheKandy, 2018-02-02:
I am on the same team as NeonOctopus, and I would like to add more to the issue. for an Autonomous program, after initializing our program, the DS phone greys out after 7-12 seconds, and the robot jerks backward a few inches. However, for a TeleOp program, the program resets after about 20 seconds and the robot does not move.

gearsincorg, 2018-02-02:
The High Ping times seem to indicate there is unintended network activity
on the phones.

Have you "forgotten" all the known WiFi connections?
If the phone is scanning the networks in the background it may cause the
disconnects and high ping times.

With your Blocks Program, can you verify that there are no "disconnected"
blocks just hanging around off-screen somewhere.
These will run even if they are not part of your program.

Phil.
​

NeonOctopus, 2018-02-04:
We have forgotten all of the WiFi networks on both phones. Both programs, the TeleOp and the autonomous have no loose strings of code. We have tried everything, but the problem has not stopped. The issue has been plaguing us for the past few competitions. If you could find any other solution, it would be very helpful.