ftctechnh/ftc_app #517: Duplicated OpMode Registration Error

CandyZQ, 2018-02-08:
One team met duplicated OpMode Registration error and asked for help in our group chat. We successfully reproduced that error by doing:

1. Create a class **ServoAdjust** and register it as TeleOp
- @TeleOp(name = "Servo Adjust",group = "Test")
- public class ServoAdjust extends LinearOpMode{
- ...}
2. Install RobotController
3. Rename the class to **Servo_Adjust** and reinstall the RobotController with instant run disabled
- @TeleOp(name = "Servo Adjust",group = "Test")
- public class Servo_Adjust extends LinearOpMode{
- ...}
4. The error shows: **"Error: An OpMode with the name 'Servo Adjust' is already registered; ignoring duplicated opmode"**
5. Change class name back to **ServoAdjust**, the error message disappears

Also, when I read the log file, I found some other interesting things. 
1. **"CS Adjust"**, which I annotated as @Disabled, was still registered. It also showed up in the list of opMode on DriverStation.
2. **"Vuforia Tracking Demo"** is a class I deleted weeks ago. It was still registered and showed up in the opMode list.

When the error was produced, OpModes could still be chosen and run in 3-4 seconds. After this period of time, none buttons(OpModes selection and Init) can be used. If I chose and started a Vuforia-related OpMode without connecting hardware, the program could still be successfully run. Another team, however, reported that although they could run the OpMode in a short period of time with hardware connected. Since I do not have Rev Expansion Hub with me now, I cannot test.   

Since the release note for v3.6 says that v3.6 fixes Block Development Tools, which we do not use, we are still using v3.5. I am not sure if v3.6 fixes this problem. 

Part of the log file:
_02-08 02:09:38.787 10896 10962 V RobotCore: Robot Status: stopped, starting robot
02-08 02:09:38.789 10896 10962 V Robocol : EventLoopManager.start()
02-08 02:09:38.789 10896 10962 V RobotCore: Getting network assistant of type: WIFIDIRECT
02-08 02:09:38.789 10896 10962 V Robocol : Spoofing a Network Connection event...
02-08 02:09:38.793 10896 10896 D WifiDirect: cannot create group, does group already exist?
02-08 02:09:38.798 10896 10962 V ThreadPool: executive service EventLoop(0x055fc62e) is terminated
02-08 02:09:38.798 10896 10962 V RobotCore: EventLoopManager state is STOPPED
02-08 02:09:38.799 10896 10962 V Robocol : eventLoop=FtcEventLoop
02-08 02:09:38.799 10896 10962 V RobotCore: EventLoopManager state is INIT
02-08 02:09:38.799 10896 10962 V RobotCore: Robot Status: initializing, starting robot
02-08 02:09:38.799 10896 10962 I FtcEventLoop: ======= INIT START =======
02-08 02:09:38.800 10896 10962 V OpmodeRegistration: registered {DefaultOpMode} as {$Stop$Robot$}
02-08 02:09:38.807 10896 10962 V FileBasedLock: locked /storage/emulated/0/FIRST/blocks/projectslock/lock.dat
02-08 02:09:38.811 10896 10962 V ThreadPool: container(0x0ebfaacc: default threadpool) added id=338 TID=0 count=2
02-08 02:09:38.812 10896 10974 V RobotCore: thread: 'SetupRunnable.run()' starting...
02-08 02:09:38.820 10896 10974 D Robocol : RobocolDatagramSocket listening addr=/192.168.49.1:20884 cbRec=65520 cbSend=81920 msRecTO=300
02-08 02:09:38.821 10896 10974 V Robocol : RecvLoopRunnable created
02-08 02:09:38.822 10896 10974 V PeerDiscovery: Starting peer discovery
02-08 02:09:38.822 10896 10974 V PeerDiscovery: No need for peer discovery, we are the peer discovery device
02-08 02:09:38.822 10896 10974 V RobotCore: Setup complete
02-08 02:09:38.822 10896 10974 V RobotCore: thread: ...terminating 'SetupRunnable.run()'
02-08 02:09:38.824 10896 10962 V FileBasedLock: unlocking /storage/emulated/0/FIRST/blocks/projectslock/lock.dat
02-08 02:09:38.829 10896 10962 V OpmodeRegistration: registered {SensorBNO055IMUCalibration} as {Sensor: BNO055 IMU Calibration}
02-08 02:09:38.829 10896 10962 V OpmodeRegistration: registered {TestVision} as {Vision Test}
02-08 02:09:38.830 10896 10962 V OpmodeRegistration: registered {Servo_Adjust} as {Servo Adjust}
02-08 02:09:38.830 10896 10962 V OpmodeRegistration: registered {TestAuto} as {Aut}
02-08 02:09:38.830 10896 10962 V OpmodeRegistration: registered {TestTeleOp} as {TestOp}
02-08 02:09:38.830 10896 10962 W RegisteredOpModes: configuration error: An OpMode with the name 'Servo Adjust' is already registered; ignoring duplicate opmode
02-08 02:09:38.831 10896 10962 V OpmodeRegistration: registered {CSAdjust} as {CS Adjust}
02-08 02:09:38.831 10896 10962 V OpmodeRegistration: registered {TeleopOpmode} as {Vuforia Tracking Demo}
02-08 02:09:38.841 10896 10977 V RobotCore: thread: 'RecvLoopRunnable.run()' starting...
02-08 02:09:38.873 10896 10962 V HardwareFactory: createHardwareMap()
02-08 02:09:38.873 10896 10962 V WifiDirectInviteMonitor: clearUILynxModule()
02-08 02:09:38.874 10896 10896 I BatteryChecker: percent remaining: 54 is charging: true
02-08 02:09:38.880 10896 10962 I RobotCore: ========= Device Information ===================================================
02-08 02:09:38.880 10896 10962 I RobotCore: Type                                               Name                           Connection
02-08 02:09:38.880 10896 10962 I FtcEventLoop: ======= INIT FINISH =======
02-08 02:09:38.880 10896 10962 V RobotCore: EventLoopManager state is RUNNING
02-08 02:09:38.900 10896 10962 V ThreadPool: container(0x0e7c2d43: SoundPlayer) added id=343 TID=0 count=1
02-08 02:09:38.900 10896 10962 V ThreadPool: container(0x0e7c2d43: SoundPlayer) added id=344 TID=0 count=2
02-08 02:09:38.901 10896 10962 V RobotCore: Robot Status: running, starting robot
02-08 02:09:38.902 10896 10962 V RobotCore: Robot Status: running
02-08 02:09:38.902 10896 10962 V ThreadPool: container(0x0fc45c3e: executorEventLoop) added id=345 TID=0 count=1
02-08 02:09:38.904 10896 10962 V RobotCore: thread: ...terminating 'RobotSetupRunnable.run()'_

[robotControllerLog.txt](https://github.com/ftctechnh/ftc_app/files/1707359/robotControllerLog.txt)

NoahAndrews, 2018-02-08:
There's actually been [some discussion](https://github.com/OpenFTC/OpenRC/issues/50) and a [Pull request](https://github.com/OpenFTC/OpenRC/pull/64) for this on the OpenRC repository that I maintain. I haven't had a chance to look at it, but the PR works by doing annnotation processing at compile time.

qwertychouskie, 2018-02-08:
Try clearing Android Studio's caches, see https://stackoverflow.com/questions/23025433/how-to-clear-gradle-cache/43245885#43245885

CandyZQ, 2018-02-09:
@qwertychouskie I think it doesn't work. I also tried to invalidate caches and restart, but it didn't help either.

qwertychouskie, 2018-02-10:
Try Build->Clean Project.

CandyZQ, 2018-02-10:
@qwertychouskie That works! Thank you very much. But it is strange that I have not met similar problems in past years.

closed 2018-04-16 by cmacfarl

qwertychouskie, 2018-04-16:
@cmacfarl This should probably be re-opened, as although there is a workaround, the issue still exists.

cmacfarl, 2018-04-16:
@qwertychouskie Hmm, the root cause appears to be that a class was renamed, and Android Studio and/or Gradle didn't prune the old class from the resultant dex file.  The way the annotation processing works is that the RC scans all the classes in the apk looking for that annotation.    If there are two classes that map to the same display name, then the error is thrown.   There's not a lot we can do about the way Android Studio is packaging the apk.  Would putting the workaround in the error message make things any better?

qwertychouskie, 2018-05-24:
@cmacfarl That would be a good stop-gap measure, though a bug report should be filed for Android Studio so the problem can be fixed.

whyiamafool, 2018-09-02:
I have this problem myself as well. Doing Build -> Clean Project does not resolve the problem for me.