ftctechnh/ftc_app #520: Driver Station fails on Android 7.1.1

IAmTheVex, 2018-02-16:
Hello! 

After we updated our driver station phone to android 7.1.1 (actually... the phone did it himself...) we can't open the app anymore! It crashes right away! Tried checking with logcat but the app is locked :( (might be a good idea to opensource it.... just sayn'..). We also tried uninstalling and reinstalling the app.... rebooting the phone. 

Thanks!

NoahAndrews, 2018-02-16:
Like the robot controller, the driver station saves a log file to /sdcard on the phone. I haven't tried reproducing yet, but it would be great if you'd provide the log.

cmacfarl, 2018-02-16:
I don't quite know what you mean by the app is locked.   If you have the phone connected to a laptop directly via usb, or via adb wireless, the log should show up in logcat.  The log itself is in /sdcard/FIRST on the phone.

If you want open source, there's always OpenFTC.  Although please do remember that's been ruled illegal for use in competition by the GDC.

NoahAndrews, 2018-02-16:
What @IAmTheVex likely meant by "locked" is that the driver station app is not debuggable, which means that you can't neatly filter its Logcat output in Android Studio like you can with a debuggable app. Also, log files are saved to /sdcard/, and not in the FIRST folder.

Note that OpenFTC does not provide an open-source Driver Station.

cmacfarl, 2018-02-16:
Oh right is Noah on all counts.  

@IAmTheVex do you get a Not Responding dialog on the driver station at any point?

IAmTheVex, 2018-02-17:
When I tap the icon to lunch the app, it crashes right away. We get the app is not responding dialog at that point. By "locked" I mean that the app is not debuggable... I can't filter the data from logcat... I can adb logcat | grep.... But what's the point of that? 😣

IAmTheVex, 2018-02-17:
I will come back with the log file soon :smile:

IAmTheVex, 2018-02-17:
I know about openftc.. I actually tried it.. But why don't just release the source? I mean... FIRST should encourage open source software and instead... We can't get the source for the app... I actually decompiled and modified the app to change the ui to match our needs.. But found out that at the competition they will "clean up" the driver station phone and install a new version of the app 🤔

IAmTheVex, 2018-02-18:
Hi! Sorry for replying sooo late :) I found the problem... is not with the app itself :( i'm sorry! we had the external sdcard mounted as the app root and somehow it could not mount the sd... sooo :)) pulled it out, cleaned it a bit, put it back in :) Now it works!

gearsincorg, 2018-02-18:
" we had the external sdcard mounted as the app root  ".

Why?

Phil Malone
GEARS Inc.

phil.malone@mr-phil.com
Office: 301.387.2331
Mobile: 301.501.7424

closed 2020-08-26 by cmacfarl