ftctechnh/ftc_app #522: Missing documentation of analog input configuration on RC app?

tjhawkey, 2018-02-18:
Hmm, there doesn't seem to be any. Why is the MR touch sensor listed as a valid selection for an analog input?

calebsander, 2018-02-18:
Not sure what you mean by "documentation of analog input configuration." FIRST only has documentation for the Android Studio/OnBot Java SDK. There are some tutorials but no real documentation for how to use the RC app. The Modern Robotics touch sensor is a digital sensor, so you can read it through an analog port by checking whether its voltage is above or below half the reference voltage. In general, you can either configure the port as a generic analog input, which will just give you voltages, or as an SDK sensor type (like the MR touch sensor) that will interpret those voltages in a specific way.

tjhawkey, 2018-02-19:
HI Caleb, it turns out that the issue with the accelerometer is that it has
a fair bit of output series resistance (for an RC filter) and the revhub
has an input voltage divider that's pretty low resistance, so together the
signal is divided by a factor of 7 or so when it's plugged into the revhub.
I'm not confident the signals are big enough for good control, but I don't
know if it's been tried yet.

Tim

tjhawkey, 2018-02-28:
Rev confirms use of MR touch sensor with analog input as a workaround.