ftctechnh/ftc_app #528: Wifi channel says it is changing but does not actually change

kkinneer, 2018-03-23:
We became aware via other teams' posts in the forums and reddit that wifi analyzers were showing that phones were not actually changing channels.  Maybe this is common knowledge so disregard this issue if so.  Or perhaps there is some known and simple workaround and the right people haven't seen what everyone is talking about yet. 

But basically, everyone thinks they are changing the channel using the FTC app settings (including the FTAs), because the phones say that the channel is changed.  But the channel isn't really changed.

In light of this, we have been doing some testing with the Nexus 5. We had terrible ping problems and lag at the North Super Regional. An FTA tried valiantly to get it to work, but no go. They tried (or thought they tried) all channels with no improvement and we were having problems both at the practice field and in the arena. The second day and before our last match, we finally gave up with the Nexus 5 and changed back to our old ZTE phones. Instantly, problem solved.  And in our testing, the ZTE phones change channels just like they're supposed to.

With a little testing here's what we learned about the Nexus 5. We used Wifi Analyzer by VREM:
- Sometimes, when the conditions are bad for 2.4GHz, the phones will choose 5GHz when starting the Robot Controller app for the first time. At least in our environment, this is sometimes actually worse than 2.4GHz. Even though there is little 5GHz traffic, ping is sometimes consistently higher than 200. Other times it's great.
- Once the app has been started for the first time, the wifi channel can't be changed except as detailed below.
- In the Robot Controller app, the "Settings" -> "Advanced" -> "Change Wifi Channel" setting does not work.  It says the channel is changed but the channel does not change.
- The phones can choose 5GHz even when you have the Android setting: Wifi -> Advanced -> "Wif-Fi frequency band" -> "2.4GHz only" on both phones. You have no choice in the matter and the FTC Robot Controller app must not even be aware that it's not in control of the channel selection.
- There are only 3 ways we could find to get a new channel (1 and 2 choose a random channel):
1. Reboot the phones and start the Robot Controller app. In our experience, our phone randomly picked among 4 channels: 1, 6, 11, and 161 (yes 5GHz)
2. Exit the Robot Controller app, swipe it away, and cycle wifi off and then on.  Start the RC app.  Again it will randomly pick from among the same 4 channels.
3. To choose your own channel, do the following procedure in the Robot Controller app:
  a. "Settings" -> "Advanced" -> "Change Wifi Channel"
  b. Select a channel and then hit the "Wifi" button at the bottom of the screen
  c. Turn off Wifi. Then Turn on Wifi. 
  d. Hit the back button to go back to the channel screen
  e. Select your channel again. 
  f. Look at your wifi analyzer app on another phone and wait until your old channel signal disappears. Note: On the Nexus 5 only, if you do not wait long enough and immediately return to the Robot Controller main screen, it will again say that it is changing the channel, but the channel is not changed.
  g. Hit the back button to get back to your Robot Controller main screen. When your wifi connects, your analyzer app should show the new channel. It will stay at this channel even if exiting the app. Just don't reboot or cycle wifi.

We also have a Galaxy S5 and tested with that.  It acts the same except that you do not have to wait like in step f above.  If cycling Wifi is a required step, the FTA's do not know it. 

One of the posts that got us looking at this was about a Moto G4 Play.  We do not have one to test with so don't know if the issue is the same.

Our team programmer made a video here: https://www.youtube.com/watch?v=vy_X_hBUGWk

Our Nexus 5's and Galaxy S5 are updated to the latest version of Android that is available which for our phones is 6.0.1 and fully updated with Play Store updates.  We are on 3.6.

gearsincorg, 2018-03-23:
Hi

This is a great analysis of a clear problem.  (I think problem demo videos are super effective.)

I am not at all versed in the ways of WiFi direct etc, but I have noticed a related issue, and I would be interested to see if it helps the situation you are seeing at all.

I have observed that the process of chanel changing (even on the ZTE phones) goes much smoother if you FIRST unpair the DS and RC phones using the normal App Pairing screen before you change the channel.

My suspicion is that if the phones are actively paired (and communicating) the RC phone is "reluctant" to have it's chanel changed.  I found that if you change the RC chanel while paired, you need to go through a sequence of app swipes and phone restarts to ensure a reconnect.  However, if you unpair first, and then change the channel, the transition happens much more cleanly.

So, if at all possible, could you try the following to see if the behavior changes:

1) Have the phones paired and running on a set channel.
2) Go to the DS App's "Settings/Pair with Robot" pane, and select "Not paired with anyone".
     Back out of the screen to execute the unpairing.  Wait a few seconds.
3) Go to the RC app and do the chanel change.
     Does the phone let go of the channel sooner?
4) Go back to the DS App and re-pair with the RC phone.
     Is the new channel now active.

kkinneer, 2018-03-23:
Grabbed the Nexus 5 phones and did the testing you suggested.  It does let go of the channel without having to "pause" as we did before.  But you still must go into wifi settings and cycle wifi off and on or the channel change will be totally ignored.  

This explains why we did not have to wait on our Galaxy S5 for the channel to be let go.  During our testing with the S5 it was not paired with another phone. 

Yes, the DS does get the new channel when you re-pair it.  I also tried without un-pairing it and just having the DS powered off.  It works just like it is unpaired.  Still must cycle wifi but do not need to "pause" to wait for channel to be let go.

I think in a live environment, it will be easier to just wait until the channel is let go than to unpair the phones.  

In 100% of our tests when the phones are paired, if the RC successfully changes channel, the DS does also.  It seems to be the RC that has the bug.  The DC has changed to the RC channel every time.  However, if the RC fails to change the channel, it will still "say" it is changing the channel and the DS will also, but they remain on the old channel.

gearsincorg, 2018-03-23:
I believe since the RC is the master, it is the one that sets the channel
for the pair of devices.  The DS uses the channel that the RC establishes.
This is why its essential that the RC changes correctly.

For what it’s worth, I stil believe that unpairing before changing Channels
makes for a cleaner transition.

kkinneer, 2018-03-23:
Mmmm, this is a tough one.  I see your point.  So far in our testing it seems to be a sure thing if you un-pair and cycle wifi.  And nobody wants to muddy it up by having a bunch of wifi analyzers around.  

Problem is our phones don't always pair easily in a busy environment.  Sometimes we have to take them somewhere else.  In an actual "live" environment where you may be in a hurry or relying on an FTA to get it done quickly, it would need something done on both phones.  Also we would need to guarantee that everyone would leave the filter on and not accidentally select another team's phone.  We had that happen to us last year when a connection request from some other team interrupted our robot right in the middle of a match.  For some reason, the RC thinks that's important and stops everything at that point (maybe that's been fixed in 3.6). 

To avoid un-pairing, we would have to see if the pause required is consistent.  In our experience, the pause is less than 10 seconds and more than 2.  We haven't further narrowed it down.

The real solution is to fix the RC app if possible.

gearsincorg, 2018-03-24:
I'm surprised to hear about your incident with an incoming pair request.

I had tested this quite a while back and determined that an incoming pair
request didn't have any impact on a running robot.  (I was concerned that
this could be an intentional exploit).

It may have changed, but if so, it wasn't intentional.

Worth retesting.
​

jdkunzeman, 2018-03-24:
Through the season I attended more than three meets and one qualifier plus some other events where long ping times were never an issue.  I was the FTA at the three meets and qualifier.  The qualifier had 25 teams and the pits, practice fields, and competition fields were all in the same room.  

Then we went to our state meet and started having significant problems with lag.  The state meet had 44 teams, two practice fields, and four competition fields all in the same large area.  There was also a school wifi network and, at times, some mobile hot spots.  Not everyone had problems.  We did hear stories later in the day that some teams knew that they needed to use ZTE phones to avoid problems, and did.  It was too late for us at that point.  (To be clear, I was not an FTA at our state meet.)

The event size and environment  seemed to cross a threshold at State.  I am a little surprised that there were problems at NSR since the fields and pit areas are separated by a good distance.  Perhaps just having four competition fields and associated queuing in the same area is enough to trigger the problems, in combination with this "channel change that doesn't actually happen" problem.  

Perhaps  others have already suggested this, but I think FTC should make a strong effort to get  a message to participating teams before the championship events:  "We strongly recommend using ZTE speed phones if you have them."

kkinneer, 2018-03-24:
I think it's a crowd thing and a channel change issue.  Or who knows.  Anyway, here's everything else I know...

We had no ping problems in any matches until State also (Iowa).  However, at State we also did not have any problems until the division finals at which point we were down to 2 fields.  What could have changed in the environment?  There were way more people in the crowd and way more kids sitting around on their phones.  And maybe once we got on a bad channel, we were stuck there because of the channel change not working.  Theoretically, there's no wifi for the crowd's phones to connect to, so why would this be an issue?  Bluetooth?  Just thinking out loud.  Or do all these wifi devices talk to each access point (each RC) a bit when they're enumerating the access points?  Why do the ZTE's work?  Because their channel changer works or maybe their wifi chips just aren't as polite as the newer phones. 

At NSR, we didn't have problems until midway through the first day, then both at the practice fields and in the arena.  Then we even had problems the next morning at the practice fields and there were not that many teams around yet.  As you mentioned, these are like completely different environments and separated by a good distance.

Also of note: At regionals we tried 3 different combinations of phones.  We started having the problems using a ZTE for the DS and Nexus 5 for the RC.  We changed to Nexus 5 for both DS and RC and the problems were worse.  We changed to ZTE  for both phones and problems solved.  When the Nexus 5 was the RC, the FTA was just pulling his hair out, because his test phones were showing way lower ping times than our phones for the "same" channel.  But of course he didn't know about the channel change issue and that our phones probably weren't actually on the same channel. 

-----

In case others have never experienced slow ping times (lag), it may be worth describing what it's like.  It basically feels like a random delay in the robot's response to the controllers.  Looking at the DS, you will see ping ranging between 100 and 900 at times but not all the time.   How this plays out in practice is that the driver would, for instance, start the robot going forward and then let go of the stick, but the robot would continue for a half second.  Or when it's bad, every single thing the driver does is delayed in getting to the robot.  The log file is clean, no errors of any kind reported. 

If you take your phones and robot to another environment (like a hotel room in our case) it will drive just fine.

qwertychouskie, 2019-02-11:
This is not directly related to this bug, but it may be useful to teams so I'll mention it here.

At worlds in our 2016-17 season we were having horrible connection issues at first.  The robot phone was mounted perpendicular to the back wall of the robot on a sliding tray.  Just changing the phone mounting so that it was flat to the back of the robot, screen facing the outside of the robot, rather than sticking a bit into the robot, made a huge connection difference for us.  There wasn't even that much metal to block signals, but apparently there was enough to affect things.

My advice to teams with connection issues:  Mount your phone on the outside if the robot, with it mounted to plastic/plexiglass.  This could also explain why some teams have a harder time on 5GHz despite less congestion, since 5GHz signals penetrate less than 2.4GHz signals.

The phone config that worked well (top middle of pic): 
[image]

For our 2017-18 season we used an Otterbox Holster Clip the the area where the camera goes cut out, and two holes drilled on the back to mount a flat plate to it.

This season we are using Polycarbonate bent by heating it with a heat gun (bottom left):
[image]