ftctechnh/ftc_app #529: Phones disconnect due to timeout while waiting for start

GabrielMajeri, 2018-03-26:
We recently updated from ZTE Speeds to Moto G4s and we are having some issues. The phones are unusable for use on the robot because they randomly disconnect.

I will also mention our phones have been updated to Android 7.0 (we saw the warning on the forums too late). We are using version 3.7 of the app, which received a patch for this 7.0.

## Steps to reproduce
What happens:
1. Select an OpMode, autonomous or tele-op.
2. Press INIT.
3. After a few (from 5 to 20) seconds, the phones will disconnect, then immediately reconnect, but the OpMode will be stopped.

Sometimes this also happens after pressing START, with the phones disconnecting while we are driving the robot.

## Error logs
[robotControllerLog.txt](https://github.com/ftctechnh/ftc_app/files/1847926/robotControllerLog.txt)
[driverStationLog.txt](https://github.com/ftctechnh/ftc_app/files/1847972/driverStationLog.txt)

The most relevant lines in the RC log are as follows:

* This is where we initialize a simple, empty OpMode ("SanityTest") that just calls `waitForStart()`.

```
03-26 14:23:20.258 10252 10363 I RobotCore: Attempting to switch to op mode Sanity Test
03-26 14:23:20.262 10252 10363 V ThreadPool: container(0x049d6add: LinearOpMode) added id=499 TID=0 count=1
03-26 14:23:20.264 10252 10400 V RobotCore: thread: 'LinearOpMode main' starting...
03-26 14:23:20.300 10252 10367 V Robocol : sending CMD_NOTIFY_INIT_OP_MODE(229), attempt: 0
```

* Nothing is wrong, phone looks fine, ping is good, then a few seconds later:

```
03-26 14:23:46.580 10252 10367 V Robocol : issuing peerDisconnected(): lastRecvPacket=2.036 s
03-26 14:23:46.581 10252 10367 V RobotCore: Network: active, disconnected
03-26 14:23:46.581 10252 10367 I RobotCore: Lost connection while running op mode: Sanity Test
```

We've noticed that in other tests that the `issuing peerDisconnected()` line is the one that shows up right when the phones stop working.

gearsincorg, 2018-03-26:
Hmm.  That looks like the same symptoms that we detected back with rev 3.4, and corrected in 3.5.

In that case the DS was getting periods were it would not "see" the incoming "keep alive" messages, so it assumed the RC was offline, so it forced it to stop a running opmode.  In that case the lastRecvPacket=2.036 s message was reported on the Driver Station.  It was an erronious condition that didn't require stopping the OPmode as the downlink was still active.

If it's now also being seen on the RC, then that's a harder problem to resolve as any loss of down traffic definitely warrants stopping the opmode for safety reasons.

In reviewing the logs, I see several different instances of the timeout occuring on the RC.  One actually causes a timeout exception:

03-26 14:23:14.843 10252 10352 E Robocol : exception SocketTimeoutException(Receive timed out): no packet received [java.net.PlainDatagramSocketImpl.receive0(Native Method)]

It really does appear that your phones are seeing comms blackouts.

When you say that you observe disconnect, then immediately reconnect.. what does it actually look like on the phones?  Any chance you could do a quick side/side video of the phones exhibiting this behavior?

GabrielMajeri, 2018-03-26:
We've done some more testing with an Samsung Galaxy S5 and we discovered something interesting: when the S5 is the Driver Station, and one of the G4s is the Robot Controller, no disconnections happen.

The Galaxy S5 (DS) is running Android 5.1.1, and the Motorola (RC) is the phone which created / hosts the Wi-Fi group.

<hr/>

When it comes to "disconnect and immediately reconnect", it happens like this:
1. The app looks OK, the GUI is working, showing the phones are connected, ping is good.
2. Then the phones disconnect, the GUI is *disabled* / *washed out*, as if the phones were disconnected. Instead of the Wi-Fi Direct network name, it shows "*Disconnected*".
3. Then the phones reconnect through Wi-Fi / re-pair, things are good, but our op-mode has been stopped.

The app never stops working or gives any errors.

As for a video, I will try to record as soon as I can. Please do tell us if we can provide any more info to help solve this issue.

cmacfarl, 2018-03-26:
Is it possible that Nougat added a battery saving feature that is disrupting the communication between phones?   Does the problem persist if you turn off battery optimizations for the robot controller and driver station?  Settings->Battery->Battery optimization->All Apps...

[image]

NoahAndrews, 2018-03-26:
Great point Craig. Nougat did greatly extend the scenarios where Doze mode can be activated.

GabrielMajeri, 2018-03-27:
@cmacfarl Thank you for your suggestion, however we already did that and it did not help:

[image]

<hr/>

@NoahAndrews We did not turn off the screen or anything. In fact, sometimes this happens in tele-op, *while we are driving*. Whatever the issue is, it happens even when there should be some activity on the phones.

<hr/>

We've also recorded a video of the phones disconnecting. The first half of the video shows the phone disconnecting after pressing INIT and waiting. The second half shows our autonomy disconnecting after waiting 10s after pressing START.

[issue.mp4.zip](https://github.com/ftctechnh/ftc_app/files/1851162/issue.mp4.zip)

cmacfarl, 2018-03-27:
The video is interesting.  My hunch is that this has to be Nougat related or we'd have seen this on more phones earlier on.   I'll see if I can get my hands on a Nougat phone or two and attempt to repro this.

LucaSeri, 2018-04-04:
Any new updates on the issue?