ftctechnh/ftc_app #531: The end of the autonomous timer does not stop motors

KadenGordon, 2018-04-06:
When the automatic 30 second timer for autonomous mode runs out, it closes the USB hub. However (at least with the REV hubs), it does not set motor powers to 0. This makes it so a robot continuously runs at whatever speed it was previously set to. It takes about 10 seconds to then crash the app, and there the REV hub will see the disconnection and stop the motors. It would be nice if the app would automatically set all motor powers to 0 before closing the connection.

gearsincorg, 2018-04-06:
What you are describing is clearly NOT the normal behavior of an app in autonomous.
So, you must have something out of the ordinary in your setup.

Please give us some supporting information so we can help diagnose the problem.
1) What type of robot hardware are you using? Modern Robotics, REV Expansion or mixed.
2) What type of phone are you using? ZTE, Moto G2, Motor 4 Play or other.
3) How did you transfer the FTC App to your Phone? Android Studio Download, Play Store download, Other?
4) CRITICAL !!!! If you used Android Studio to deploy your app... DID YOU TURN OFF AUTO RUN? You MUST do this. !!!!!!!!!!!!!!!!!!!!!!!
File / Settings / Build-Execution-Deployment / Instant Run (uncheck)
5) What is the version of the RC App? Use the menu to show the About screen. Rev 3.6 is the latest rev, but it could be 3.1 or 2.62 or even 2.4 if it's from early last year.
6) Is this a problem that has started since you made a change to the system? What was that change?
7) If you are having crashes or disconnects or anomalous behavior, be ready to provide a copy of you log file (found on phone) to help in debugging.

KadenGordon, 2018-04-07:
1. All REV
2. Nexus 5
3. Android Studio
4. It is off
5. 3.5
6. It has always done this in my experience
7. There is no abnormal crashes or disconnects other than the error that says "OpMode stuck in stop(). Restarting robot controller app". It then restarts the app and we are able to pick another OpMode. It takes about 30-40 seconds for all this to happen. 
[image]

calebsander, 2018-04-07:
The `stuck in stop()` is the real issue. Something is preventing the `OpMode` from ending when requested. If you are using a `LinearOpMode`, you probably have a loop that is not running only while `opModeIsActive()` returns `true`.

calebsander, 2018-04-07:
Yes, that is the problem. You should include `opModeIsActive() &&` in front of all your `while` loop conditions so that if the `OpMode` is stopped, `runOpMode()` will terminate. This will also add a call to `idle()` each loop, which is good because it prevents your code from looping faster than new sensor inputs, etc. can be read.

gearsincorg, 2018-04-07:
Yes...

You have a lot of while loops that have no test to ensure that your opmode
exits correctly.

eg:

while (findGlyphTime.seconds() < 3.5)  {

should be

while (opModeIsActive() && (findGlyphTime.seconds() < 3.5))  {

without this added clause, when the auto mode is ended, your code may not
exit as it should, so the Watchdog timer flags a "stuck" error and then
restarts the app.
The Watchdog timer is for two seconds, and then the restart may take 5-10
seconds.
These conditions are what you are seeing as the REV hub being "closed".
They will not happen if you provide a way for your while loops to exit as
requested..

I also suspect that your driving code may also be missing these safety
tests.
eg: you have many calls that look like some variation of:

robot.drive.forward(robot.drive.DRIVE_INTO_GLYPH_PIT_SPEED,
robot.drive.DRIVE_INTO_GLYPH_PIT_DISTANCE);
or
robot.drive.backward(robot.drive.MAX_SPEED,
robot.drive.DRIVE_INTO_GLYPH_PIT_DISTANCE - 4);
or
robot.leftGyro(robot.drive.MAX_SPEED, -90);

Each of these are probably looping internally waiting for the condition to
be true.
I would check to ensure that these calls (and the other ones like them)
also have a  opModeIsActive() in their loops.

note:  I'm surprised you robot actually passes field inspection.  The basic
auto-teleop transition test should have been flagged.

Phil.
​

KadenGordon, 2018-04-10:
We implemented opModeIsActive() into our code. It works so far. I'll reopen this if something else happens that is obviously not caused by the same thing

closed 2018-04-10 by KadenGordon