ftctechnh/ftc_app #533: Cannot Resolve Symbol from Basic LinearOpMode

Stiffy2000, 2018-04-08:
So, I have imported the Basic Linear OpMode from the samples directory into my teamcode folder. I build and I get "Cannot resolve symbol". The build goes fine, but in the import statements, the com.qualcomm.robotcontroller folder is not being found, even though they are in the dependencies.

NoahAndrews, 2018-04-09:
Try choosing `File > Invalidate Caches / Restart` in Android Studio and try again

gearsincorg, 2018-04-09:
These symptoms often result from a miss-configured or recently updated development environment.

How did you get to this point?    That is:  
Are you starting out with a fresh install of Android Studio, and FTC-SDK, or are you upgrading either of these components from an earlier installation?

Give us a bit of history so we can give you appropriate guidance.

Phil.

NoahAndrews, 2018-04-17:
I just downloaded your 3.7-based repo, and successfully built it with no issues. 

What happens if you click the Build (hammer) button next to the dropdown that says TeamCode? Does that fail too, or is it purely an issue with Android Studio not knowing what's up?

FTC6567RoboRaiders, 2018-04-17:
Noah - 

Thanks for the reply.

Tried the Build button and that did not resolve the issue.  Tried the Gradle Sync as well and still having the issue. 

Looking through the Sync output, I see two warnings.  

C:/Users/<userid>/StudioProjects/RoboRaiders_2017_2018_ftc37_app |  
TeamCode/build.gradle |  
Configuration 'compile' is obsolete and has been replaced with 'implementation' and 'api'. It will be removed at the end of 2018. For more information see: http://d.android.com/r/tools/update-dependency-configurations.html |  
FtcRobotController/build.gradle |  
Configuration 'compile' is obsolete and has been replaced with 'implementation' and 'api'. It will be removed at the end of 2018. For more information see: http://d.android.com/r/tools/update-dependency-configurations.html |  

Here is a screen grab if that is easier to see:
[image]

FTC6567RoboRaiders, 2018-04-17:
So I did a google search with the following keywords "android studio 3.1.1 cannot resolve symbol" (without the quotes).  The top hit for me is https://stackoverflow.com/questions/49523302/android-studio-3-1-cannot-resolve-symbol-themes-widget-attr-etc

I tried the first (top) answer and magically it worked.  I have no idea what that really did or didn't do.  

So (for now) I would consider this problem for me closed/resolved.

Thanks all for the help.

Stiffy2000, 2018-04-30:
I found the fix some time ago. Sorry for not alerting you all previously.
Visit [this link](https://emaildebtforgiveness.me/) to learn more.

closed 2018-04-30 by Stiffy2000

NoahAndrews, 2018-05-01:
I think that's the wrong link.

Stiffy2000, 2018-05-02:
Ok, to fix this, I changed the import statements. Look in the teamcode
folder and one of the files — say Elevator — and use the import statements
from there.
-- 
Best Regards, Sambhav Saggi CEO, Chairman and Founder Of Seeker S3

whyiamafool, 2018-07-15:
FTC6567RoboRaiders's solution worked! Thanks so much!

RaduSur, 2019-11-28:
Did you perhaps deleted the ElapsedTime statement ? that`s why i gor the error at least