ftctechnh/ftc_app #535: WiFi connection info display and logging

AlecHub, 2018-04-20:
Request Driver Station to display WiFi signal level, link speed, and channel number.  Useful for WiFi diagnostic purposes and to ensure team is using the assigned channel.  Also request Robot Controller to log changes and/or anomalies in signal level, link speed, and channel number.

Relevant JDK APIs:
```java
import android.net.wifi.WifiInfo;
import android.net.wifi.WifiManager;

WifiInfo wifiInfo = wifiManager.getConnectionInfo();

int rssi = wifiInfo.getRssi();  // signal strength in dBm
int frequency = wifiInfo.getFrequency();  // connection frequency
int linkSpeed = wifiInfo.getLinkSpeed();  // link speed

// convert rssi to signal level from 0 to 10

int signalLevel = wifiManager.calculateSignalLevel(rssi, 11); 

// convert frequency to 802.11 channel number and band

double band;
int channel;

if (frequency >= 2412 && frequency <= 2484) 
{
   band = 2.4;
   channel = (frequency - 2412) / 5;
}
else if (frequency >= 4915 && frequency <= 5865)
{
   band = 5.0;
   channel = 0; // TODO
} 
else
{
   band = -1.0;
   channel = -1;
}
```

FROGbots-4634, 2018-06-13:
@AlecHub This would only work if FTC changes to use soft-AP instead of WiFi-Direct.

If `wifiManager.getConnectionInfo();` is called when connected via WiFi-Direct, it will not return any useful information.

Additionally, `WifiP2pManager` does not provide a `getConnectionInfo()` method, so I'm afraid this is not possible.

AlecHub, 2018-06-14:
Thanks FROGBots...  I didn't realize that `WifiManager` and `WifiInfo` doesn't apply to WiFi-Direct connections.  Good catch!