ftctechnh/ftc_app #538: Compiler Error

ToxicObsidian, 2018-05-24:
Hi Contributors:
    I'm a FIRSETer from FTC Team 1820 (China).
    1.Recently we've got into some trouble when we attempted to compile our java files.The compiler returned that there's an error occurred in our file "CpAOpMode.java" , with the 23th line.But we have deleted this file some time ago , we've tried to recreate the file and rewrite it , but the compiler still returned error.
    2.Another compiler also seems wrong , we have another installation on another mobile phone . It seems normal this time , but once we edited our file , and compiled it , the compiler returned FAILED without any errors . We are so anxious because we don't have enough time now , the competition falls on 2nd June - 3rd June , we are now troubling by the compilers.
    Though we have installed FTC_app v3.6 , we didn't find that you have fixed these problems( Version 3.7 changes include : Interim fix to allow FTC apps to run on Android Nougat (7.x) devices. ).

ToxicObsidian, 2018-05-24:
You see I'm a Chinese senior student , so i'm not so good in English

dmssargent, 2018-05-24:
Logs would be helpful for both cases. I'm assuming you are using OnBotJava. You can download the logs from the Manage screen.

ToxicObsidian, 2018-05-25:
Emm... our teammates have already solved this problem by formatting the entire phone and reinstall the RobotController . Anyway , we are very appreciate for your reply.

closed 2018-05-25 by ToxicObsidian