ftctechnh/ftc_app #539: How can i use UltraSonic sensor

gunrh1231, 2018-05-29:
hey,
I want to use an ultrasonic sensor that is not modern robotics ultrasonic sensor, like this one [here ](https://www.seeedstudio.com/Grove-Ultrasonic-Ranger-p-960.html) https://www.seeedstudio.com/Grove-Ultrasonic-Ranger-p-960.html or this [one](https://www.tetrixrobotics.com/Ultrasonic-Sensor-Pack) https://www.tetrixrobotics.com/Ultrasonic-Sensor-Pack.
I know this two sensores are digital.
how can this be done?
I did not understand how I address him in the hub in the digital sensors, as a digital device?
and if so in the code as a digital channel?
I do not know please help 

Is this code ok? 

package org.firstinspires.ftc.teamcode;

import com.qualcomm.robotcore.hardware.UltrasonicSensor;
import com.qualcomm.robotcore.eventloop.opmode.Autonomous;
import com.qualcomm.robotcore.eventloop.opmode.LinearOpMode;

@Autonomous(name = "Sensor: UltrasonicSensor", group = "Sensor")

public class UltrasonicSensor extends LinearOpMode {

    UltrasonicSensor rangeSensor;

    @Override public void runOpMode() {

        rangeSensor = hardwareMap.get(UltrasonicSensor.class, "sensor_range");

        waitForStart();

        while (opModeIsActive()) {
            telemetry.addData("cm", "%.2f cm", rangeSensor.getUltrasonicLevel());
            telemetry.update();
        }
    }
}

[image]

AlecHub, 2018-06-12:
The Grove Ultrasonic Sensor has a PWM interface and the Tetrix Ultrasonic Sensor has a serial interface.  Neither sensor is compatible with the current sensor interface options for FTC, namely I2C, Analog, and Digital.  You need to use a ultrasonic sensor that has either a I2C or Analog interface.

FROGbots-4634, 2018-06-14:
I would recommend you check out this line of sensors: https://www.maxbotix.com/product-category/i2cxl-maxsonar-ez-products

closed 2020-08-26 by cmacfarl