ftctechnh/ftc_app #546: Configuration 'compile' is obsolete and has been replaced with 'implementation' and 'api'. It will be removed at the end of 2018. Error

whyiamafool, 2018-07-16:
Configuration 'compile' is obsolete and has been replaced with 'implementation' and 'api'.
It will be removed at the end of 2018. For more information see: http://d.android.com/r/tools/update-dependency-configurations.html

This warning is shown as a Gradle Sync Issues for TeamCode/build.gradle and FtcRobotController/build.gradle

How do I fix this?

NoahAndrews, 2018-07-16:
Ignore it, it will be gone in the software for next season.

whyiamafool, 2018-07-16:
@NoahAndrews, when will the ftc_app for next season release?

NoahAndrews, 2018-07-16:
I'm afraid I don't speak for FIRST.

dadosch, 2018-09-15:
In FtcRobotController's and TeamCode's gradle file (build.gradle @ TeamCode and build.release.gradle @ FtcRobotController) change "compile" to "implementation". It should work, but no warranties :smile:

JBYoshi, 2018-09-15:
This should probably be fixed by FIRST as well.

NoahAndrews, 2018-09-15:
Thanks for bringing this back up. Turns out we only fixed it for the internal build.

rwoodley, 2018-10-25:
Can you tell us what the fix is so we can implement it now? Or point us to a commit?

dmssargent, 2018-10-25:
Take a look at the current alpha: https://github.com/ftctechnh/ftc_app/commit/58d723a1d11a1c03ffb3a30f4a38c3601eac61d8#diff-a1a86c17a464097b827eab71781245d2R2

closed 2020-08-26 by cmacfarl