ftctechnh/ftc_app #549: "Configuring Wi-Fi Direct..."

mihai02t, 2018-08-03:
Hello! I have some weird things happening with the Driver Station app. I'm using Android Studio.
When I pair the Robot Controller phone with the Driver Station phone through Wi-Fi Direct, they connect successully. However, when I try to open the Driver Station app, it gives me an error with "Configuring Wi-Fi Direct...", stays like that for 5 seconds, the app restarts and the loop continues. I can't even reach the main activity.
However, when the devices are not connected to each other, the app opens fine. So this issue seems to happen only when the Driver Station device is connected to other device through Wi-Fi Direct. Both the Driver Station and the Robot Controller apps are the same version.
I tried swapping the phones with other ones, same issue. I think it is something wrong with my Robot Controller app. I attached a log file from the Driver Station device. Thanks!

[driverStationLog.txt](https://github.com/ftctechnh/ftc_app/files/2257877/driverStationLog.txt)

NoahAndrews, 2018-08-03:
Does it work if you pair the devices through the driver station app, rather than through the system Wifi Direct screens?

mihai02t, 2018-08-03:
@NoahAndrews No, unfortunately. It doesn't want to connect.

gearsincorg, 2018-09-14:
As a general rule, the recommendation is to NOT connect the phones using the native WiFi direct screens.  ONLY use the pairing mechanism that the APPs offer.   In fact it's recommended that you FORGET all WiFi direct groups.

The apps are doing more and more to help streamline the various connections, so if you do your own connection management it may be counter to what the apps are doing.

I'm not saying that this is the problem in your case, but any request for help is clouded by the fact that you are making connections behind the scenes and it may cause symptoms that aren't easily recognized.

Since you have your phones in an unknown state, you need to get to a clean slate.
I would recommend you do each one of these steps.

1) Tell BOTH phones to forget all your WiFi and WiFi Direct connections and groups.  Do this through the phone's wireless setting page.
2) Open the DS app and go to the settings/pairing screen, and select NONE as your option.
3) Swipe off all the apps on the DS and RC phones.
4) Do a full reset or power down on both phones.  Then start them up again.

5) Start the RC app.  Let it get all the way up.  
6) Do a Settings/Self-inspect and make sure all the items are green.

7) Start the DS app.  Let it get all the way up
8) Do a Settings/Self-inspect and make sure all the items are green.

9) Go to Settings/Pairing  The RC phone should be listed.
    If it's not, turn off the filter, just in case your phone names are incorrect
    Select the desired phone and back out of the screen.

10) The invite should appear on the DS phone.  Say Yes.

11) At this point the phones should connect.  I've had times that this doesn't happen right away.  I usually swipe both apps off, and then start them again.  They always connect then.

If you run through this sequence and they don't pair, please report at which point the sequence fails.