ftctechnh/ftc_app #550: Incorrect Javadoc for Acceleration and Velocity

NoahW314, 2018-08-09:
In the Javadoc description of `Acceleration` and `Velocity,` it states that `Acceleration` "represents a double integration of `Position` over time" and that `Velocity` "represents an integration of `Position` over time."  This is incorrect.  The Javadoc should state that a double integration of `Acceleration` over time will result in `Position` and a integration of `Velocity` over time will result in `Position`.

AlecHub, 2018-08-17:
Hi Noah, you can fork the repo, make the corrections in your fork, then issue a pull request of your corrections.  I think it would be instructive for FTC students to see the fork/pull request workflow in action.

rbrott, 2018-08-17:
Except that this requires changes to the actual source which is not directly available through this repository. From `CONTRIBUTING.md`, the SDK developers seem to prefer that students submit bug reports/feature requests and leaving them to deal with the implementation.

WardBenjamin, 2018-08-17:
@AlecHub, the actual source of the JavaDoc is the documentation strings embedded in the source code, which we do not have direct access to as per @rbrott's comments. Modifying the HTML docs is incorrect and pointless since they're simply generated from the in-code notes.

AlecHub, 2018-08-17:
Hi Ward, I didn't realize that the HTML is generated. So yes, it would be incorrect to modify the HTML in this context.

closed 2018-08-21 by ftctechnh

dmssargent, 2018-08-21:
This issue will be fixed next SDK release.